ArduPilot/ardupilot #22780: SmartRTL causing Internal Error 100000

andrefreitas97, 2023-01-30:
## Bug report

**Issue details**

If I engage SmartRTL with a switch 3 consecutive times while the drone is disarmed I get Internal Error 100000 at the third time. I believe this should be some type of bug.

**Version**
4.3.3

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Kakute H7

shivam969619, 2023-02-04:
I am new here. Can i take this issue.

prathamesh7255, 2023-02-04:
@andrefreitas97 when it shows the error for the third time does it take time of 10 seconds or does it show the error immediately?

prathamesh7255, 2023-02-04:
> I am new here. Can I take this issue.

Hello Shivam, I would recommend you start with good first issues.

prathamesh7255, 2023-02-05:
> @andrefreitas97 when it shows the error for the third time does it take time of 10 seconds or does it show the error immediately?
@rmackay9 if you know, please help.

andrefreitas97, 2023-02-06:
@prathamesh7255 the error shows immediately after the third time I engage SmartRTL

prathamesh7255, 2023-02-06:
@andrefreitas97 I think the issue is almost resolved but as I was not able to reproduce the issue on my GCS, would it be fine for you to have an online meeting to clear some doubts? if it's fine, you can mail me the time which suits for you. you will find my email on my profile.

prathamesh7255, 2023-02-18:
@andrefreitas97, how many total waypoints were there when engaging smart RTL, three or more?

andrefreitas97, 2023-02-18:
There were none because it had just been turned on on my desk with no gps fix

prathamesh7255, 2023-02-20:
> There were none because it had just been turned on on my desk with no gps fix

@rmackay9 sir, I think the absence of fixed GPS is causing the error.
What should we do then? as we have several choices:
1. we can consider adding a check for GPS lock before engaging SmartRTL. If the GPS lock is unavailable, we can either prevent the SmartRTL mode from being engaged or we can warn the user that SmartRTL is not recommended without a GPS lock.
2. we may also consider implementing a timeout mechanism for GPS lock acquisition, where the SmartRTL mode is automatically disengaged if the GPS lock is not acquired within a certain period of time.

rmackay9, 2023-03-09:
I think this issue is a duplicate of https://github.com/ArduPilot/ardupilot/issues/17416

rmackay9, 2023-03-09:
This PR could resolve the issue https://github.com/ArduPilot/ardupilot/pull/17419