ArduPilot/ardupilot #22849: MAV_CMD_DO_SET_ROI_LOCATION Bug

mustafa-gokce, 2023-02-07:
## Bug report

**Issue details**

As per the suggestion of the MAVLink docs here: [MAV_CMD_DO_SET_ROI_LOCATION](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_ROI_LOCATION)
When you send MAV_CMD_DO_SET_ROI_LOCATION command to the vehicle (Copter) from MAVProxy like:
```
long MAV_CMD_DO_SET_ROI_LOCATION 0 0 0 0 -353632621 1491652374 0
```
The simulated vehicle SITL instance immediately stops working.
MAVLink document suggests to latitude and longitude fields should be in 1e7 form but firmware and also [MAVProxy](https://github.com/ArduPilot/MAVProxy/blob/93ecdcf9c31b9e4aa53d94f30739267108315139/MAVProxy/modules/mavproxy_map/__init__.py#L630) implemented this feature with float numbers.

**Version**
master (and AFAIK all the versions before) has this.

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Copter

**Hardware type**
N/A

**Logs**
N/A

Puni2001, 2023-02-07:
Hey ! Can I Take this issue to fix

rmackay9, 2023-02-08:
@Puni2001, sure go for it.

One thing to check though (or perhaps @mustafa-gokce can check) is whether the command works properly if the lat, lon and alt are specified in degrees.  e.g. does this work OK?  does the copter point at the correct location?

> long MAV_CMD_DO_SET_ROI_LOCATION 0 0 0 0 -35.3632621 149.1652374 0

If the vehicle does point at the correct location then the issue is just that MAVProxy is automatically scaling the lat, lon and alt to degrees so the original command, (e.g. "long MAV_CMD_DO_SET_ROI_LOCATION 0 0 0 0 -353632621 1491652374 0") is an incorrect command to use and we may not need to actually fix anything.  Of course AP should protect against failures if they occur on real vehicles but it's not clear to me that this failure in SITL would actually happen on a real vehicle.  On a real vehicle the location might become invalid and the command might be rejected.

Puni2001, 2023-02-11:
This pull request resolves the issue with the MAV_CMD_DO_SET_ROI_LOCATION command in the simulated vehicle SITL instance.

According to the MAVLink docs, the latitude and longitude fields should be in 1e7 form, however, the firmware and also MAVProxy implemented this feature with float numbers, causing the simulated vehicle SITL instance to immediately stop working.

This pull request changes the code so that the lat, lon, and alt fields are correctly specified in 1e7 form, ensuring that the simulated vehicle SITL instance will function correctly.

Changes Made
The cmd_set_roi function was updated to convert the lat, lon fields to int and multiply them by 1e7 before sending the MAV_CMD_DO_SET_ROI_LOCATION command.

Types of changes
Bug fix

Puni2001, 2023-02-12:
@rmackay9, This pull request resolves the issue with the MAV_CMD_DO_SET_ROI_LOCATION command in the simulated vehicle SITL instance.

According to the MAVLink docs, the latitude and longitude fields should be in 1e7 form, however, the firmware and also MAVProxy implemented this feature with float numbers, causing the simulated vehicle SITL instance to immediately stop working.

This pull request changes the code so that the lat, lon, and alt fields are correctly specified in 1e7 form, ensuring that the simulated vehicle SITL instance will function correctly.

Changes Made
The cmd_set_roi function was updated to convert the lat, lon fields to int and multiply them by 1e7 before sending the MAV_CMD_DO_SET_ROI_LOCATION command.

Types of changes
Bug fix