ArduPilot/ardupilot #22886: Arduplane + CRSF = dead link after FC reboot

ABLomas, 2023-02-09:
ArduPlane V4.3.3 (d808133a) (official release)
New setup, starting from scratch, no config import.
Matek F405-WMN connected with Crossfire Nano RX using TX2/RX2 as manufacturer recommends (http://www.mateksys.com/?portfolio=f405-wmn#tab-id-4) (crossfire protocol, SERIAL6_OPTIONS ->  0, SERIAL6_PROTOCOL -> 23 ).
RC link is working after setup, can calibrate radio, see values change, failsafe detected, etc - all fine.
BUT - if i reboot FC (for example in compass configuration or Actions -> preflight reboot) - RC link is dead until complete power cycle.
RX itself i think is working fine - still green LED, goes red if i turn off TX, etc.
FC goes info failsafe (as does not see RC link) and i did not found easy way to "revive" RC link except power cycle (no reconfiguration needed - just power cycle and link is up again).

Also FC correctly reacts into failsafe condition - i can turn off TX, turn it on - RC link works, but FC reboot kills it.

Legzeg, 2023-03-09:
Absolutely same behaviour on:
- TBS Diversity Rx;
- H743 wing v3
- Serial7 (as it recommended by matek)

ABLomas, did you find the solution on your issue?

ABLomas, 2023-03-09:
> @ABLomas , did you find the solution on your issue?

Yeah, INAV =)

I know, i know - this is not what you expected, but i ran into few serious problems (which are more "think enterprise, work like enterprise" problem) with arduplane and ran away.

andyp1per, 2023-03-09:
I did put a fix for this into 4.3, but I have noticed this behaviour of late so something is not working as intended. Note that CRSFv3 makes this harder with its bootstrap protocol. What TBS firmware version are you using (because it matters).

Legzeg, 2023-03-10:
For now it`s the last firmware - 6.19
Did I have to downgrade it?

Legzeg, 2023-03-10:
TBS Crossfire Diversity Nano RX <-> H743 wing v3 (Ardupilot 4.3.4) <-> TBS Tango II Pro

andyp1per, 2023-03-10:
Ok thanks - I have a comparable setup, I will do more testing

andyp1per, 2023-03-10:
> Well, i have 6.09 firmware (not latest, for sure) and had problems described in initial post.

Yeah you need at least 6.17 for this to work

RBirdie001, 2023-06-18:
Sorry, bit confused: should it be solved now or not? I have ArduPlane 4.3.7 and Crossfire 6.19 and have exactly this problem.
If it's known not to be solved, is there any estimation?
Thanks!

andyp1per, 2023-06-19:
It should be fixed in 4.3.x. Can you explain how you reproduce the problem? What CRSF module are you using?

RBirdie001, 2023-06-20:
I'm using crossfire nano 868 receiver with firmware 6.19 connected to SERIAL1 of FC Omnibus F4 V3.
Crossfire is set to output CRSF, FC is running Arduplane 4.3.7 and SERIAL1_PROTOCOL =23.
On the SERIAL3 is connected ublox GPS @115K, on SERIAL4 (BRD_ALT_CONFIG=3) is Mavlink 2 telemetry modem at 38K.
When I **boot** by connecting power, receiver is detected independent whether RC receiver is already at the boot time outputting CRSF (radio on, link estabilished) or receiver still in FS (RC FS = no output). Then is detected after estabilishing the link.
When I **reboot** FC by command “Actions/preflight reboot shutdown” either by USB or telemetry modem at SERIAL4, receiver is the not detected and FC remains in failsafe even is constantly outputting CRSF.

All is described here: https://discuss.ardupilot.org/t/crossfire-receiver-not-detected-after-software-reboot/102350

If you need parameters, any specific test or log, I'm ready to help solving this.
Thanks!

andyp1per, 2023-06-20:
Thanks, please post your parameters - I have all that hardware so should be able to replicate easily

RBirdie001, 2023-06-20:
Here it is:
[Talon#2_single_437_baro_rngfndr_20230613.zip](https://github.com/ArduPilot/ardupilot/files/11800442/Talon.2_single_437_baro_rngfndr_20230613.zip)

RBirdie001, 2023-06-20:
I tested it on the different piece of FC and receiver of the same type and:
With the same settings the same problem (NO RC after reboot).
With default settings, only set SERIAL1_PROTOCOL =23 it's always detected even after soft reboot.

andyp1per, 2023-06-20:
So does any setting of BRD_ALT_CONFIG cause it to fail? Or only ALT 3?

RBirdie001, 2023-06-20:
I'm confused, it's more complicated: Now, after hardare reboot it fails at all software reboots with any BRD_ALT_CONFIG including 0.
If I do this sequence: reset all parameters to default, change SERIAL1_PROTOCOL =23, soft reboot, reboot, reboot... it always works.
If I then do first hard boot, it works, but then at any next soft reboot it fails again and again...
Not very smart of this behaviour :-(

andyp1per, 2023-06-20:
Is there any different CRSF version negotiated in the different cases?Sent from my iPhone

RBirdie001, 2023-06-20:
Sorry, I don't know how could I test it as I don't understand these details deep enough. Do you have any idea how to test it?

andyp1per, 2023-07-01:
The issue is CRSFv3. It is supposed to downgrade the serial link to 416k from 2Mb when it receives no telemetry for some period, but this is not happening. The RX continues talking at 2Mb and the uart is talking at 416k.

closed 2023-07-19 by rmackay9