ArduPilot/ardupilot #22900: AP_Mount: add FPV mode support for at least Siyi gimbals

rmackay9, 2023-02-11:
The Siyi A8 (and perhaps other gimbals) support "FPV" mode which means the gimbal's roll and pitch move in body-frame instead of earth-frame.  The user can still control the angle or rates of the gimbal but the movements are all relative to the body of the vehicle.

Follow: roll and pitch are in earth-frame, yaw is in body-frame
Lock: roll, pitch and yaw are all in earth-frame
FPV: roll, pitch and yaw are all in body-frame

We should add support for users to switch between FPV, Follow and Lock modes from the transmitter.  One solution is to extend the existing "Mount Lock" RC switch so that the middle position is FPV.  This will also require some changes in the AP_Mount drivers especially the AP_Mount_Siyi driver to implement this.

This request comes from [this discussion](https://discuss.ardupilot.org/t/siyi-a8-mini-4k-ai-mini-zoom-gimbal-camera-ai-identify-tracking-4k-1-1-7-inch-8mp-sony-sensor-6x-zoom/92133/154).

afonsoVale, 2024-12-02:
Hello all,

Following up on this, do you have any news on the implementation of this?

In particular, does the current version of ArduPilot support setting the SIYI A8 Mini gimbal in FPV mode? I saw there was a merged pull-request associated with this issue which looks like it may be implementing this, but I can't seem to get it working (assuming that the MIDDLE value is what sets it to FPV mode).

Thank you!

rmackay9, 2024-12-03:
Hi @afonsoVale,

I don't think we've merged a change to add support for this.  There is some working going on to add it though including as part of this PR for another gimbal with a similar feature https://github.com/ArduPilot/ardupilot/pull/28373

Nemrond, 2025-02-13:
¡Great!, hope this come soon. And one more thing, ¿will be posible to use this mini gimbal to make it rotate whole itself (independently of flight controller attitude, taking it from downside to upside mode) to hide it into the fuselage, in a kind of platform with servos (what i'm doing), to protect it from a hard land or dust? I'm not a developer, but i understand that if FPV will be implemented, it will need to receive flight controller attitude and position, also that i need to power it in the mode that will be used in the whole flight (up or downside). 

Probably not. Or maybe in case the gimbal detects a huge difference of attitude between FC, could let to a cutoff of attitude data stream from FC and got it from itself instead.

Sorry for my english..