ArduPilot/ardupilot #22927: Drone disarms immediately after takeoff item

mosheFuks, 2023-02-14:
**Issue details**
When the Arm/Disarm button is priced, the drone is armed, the console shows Takeoff but immediately disarms. Any idea what can be happening? 

**Version**
I'm using Ardupilot with AirSim

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
SITL

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2023-02-15:
@mosheFuks,

.. an onboard log would be good in order to figure out the cause of the disarm.  A random guess is the DISARM_DELAY is set to a low value (like 1) or a failsafe is triggering and because the vehicle is still landed this is triggering the disarm.

mosheFuks, 2023-02-16:
Hi @rmackay9, DISARM_DELAY was set to 0 (disabled), and all checks in ARMING_CHECKS are enabled. What happens is that immediately after doing the Arm does, the Disarm (without getting to do the takeoff as before). In AirSim, doing the Arm in stabilize mode and the takeoff in Guided mode is necessary. Can this be related? 
I've attached the logs. (I needed to change the BIN extension to txt)

[00000014.txt](https://github.com/ArduPilot/ardupilot/files/10758609/00000014.txt)

peterbarker, 2024-11-23:
All arming/disarming in this log is done via the GCS.  ie.  the GCS is disarming the vehicle.

```
2023-02-17 03:34:47.689 ARM {TimeUS : 564804000, ArmState : 0, ArmChecks : 0, Forced : 0, Method : 1}
2023-02-17 03:36:15.396 ARM {TimeUS : 652510000, ArmState : 1, ArmChecks : 33554422, Forced : 0, Method : 1}
2023-02-17 03:36:18.120 ARM {TimeUS : 655234000, ArmState : 0, ArmChecks : 0, Forced : 0, Method : 1}
2023-02-17 03:37:03.899 ARM {TimeUS : 701014000, ArmState : 1, ArmChecks : 33554422, Forced : 0, Method : 1}
2023-02-17 03:37:05.932 ARM {TimeUS : 703046000, ArmState : 0, ArmChecks : 0, Forced : 0, Method : 1}
```

method 1 is "GCS".

So I think this is a problem with whatever GPS was being used.  Please re-open if you think there's more to this.

closed 2024-11-23 by peterbarker