ArduPilot/ardupilot #22949: AP_VisuaOdom:Add single position source when using motion capture systems

htchit, 2023-02-16:
## Feature request
When I using motion capture system as position source, I found ArduCopter firmware can not estimate height accurately.There's always 5cm error higher or lower around the position it should be.

**Describe the solution you'd like**
When I using high accuracy positioning source, can I turn off other position estimator such as integration from IMU's accel or just bypass EKF to get the stable height estimate.

This is my first time to submit an issue, hope i didn't make any mistake about it.lol
**Platform**
[*] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

htchit, 2023-02-19:
I have solved this problem by add an "ekf bypass" to the InertialNav,anyway,it's not a good way to achieve the goal because I edited 2 libraries and Copter.cpp. So i would not pull this code to the main branch.If anybody wants the code,please email me.I'm willing to help!

closed 2023-02-19 by htchit

magicrub, 2023-02-19:
Welcome! Please post a Pull Request of your code, maybe we'll all learn something!