ArduPilot/ardupilot #22952: Copter: KDECAN doesn't appear to be working

KDEDirect, 2023-02-16:
**Issue details**
KDECAN doesn't appear to be working on ArduCopter firmware 4.3.0 and above. We've had quite a few customers complain about this who are excited to use the new firmware but can't because it isn't working with the latest versions. I tested KDECAN on ArduCopter 4.2.3 today and it works great.

What's odd is that the flight controller sends the message to ping the ESCs, then the flight controller seems to go into an error state because it wont connect via mavlink or arm. If I disconnect the CAN wire that connects to the flight controller and reboot, then I can arm and connect via mavlink normally.

If someone wants to troubleshoot this, please let us know and we can send some free ESCs and CAN wires your way :)

**Version**
ArduCopter 4.3.3

**Platform**
[ ] All
[ ] AntennaTracker
[ x ] Copter
[ ] Plane
[ ] Rover
[ ] Submarine

**Hardware type**
Pixhawk, CubeOrange, Pixhawk4, etc.

**Logs**
Older ArduCopter firmware transmits CAN messages after verifying ESCs are present on the bus but the new firmware only sends the first message then stops transmitting:

[image]

rmackay9, 2023-02-16:
Thanks for the report, I've added this to the Copter 4.3 issues list so it is not forgotten. https://github.com/ArduPilot/ardupilot/issues/21711

I'm probably the best person to look into this because I have two of the ESCs and I last touched the driver.

magicrub, 2023-02-17:
I've totally rewritten this whole library but don't know how to test the enumeration stuff. @KDEDirect  If I make a PR and get you a binary, can you test it for me? What hardware shall I build it for?

KDEDirect, 2023-02-17:
I have a [CubeBlack](https://firmware.ardupilot.org/Copter/stable-4.2.3/CubeBlack/) sitting on my desk right now. If you want to send a binary, I'd be happy to test it!

magicrub, 2023-02-17:
@KDEDirect I pinged you on LinkedIn. Lets chat!

KDEDirect, 2023-02-17:
The marketing department is currently out of office. You can reach our engineering team by creating a [support ticket](https://www.kdedirect.com/pages/support-ticket2) or by reaching me via email: broc.nickodemus@kdedirect.com
Thanks,
Broc

KDEDirect, 2023-03-24:
Fixed with PR https://github.com/ArduPilot/ardupilot/pull/23238