ArduPilot/ardupilot #22957: rover: Scurve produces inconsistent effects under the same turning conditions

xianglunkai, 2023-02-17:
[image]

xianglunkai, 2023-02-17:
bool s_finished = _scurve_this_leg.advance_target_along_track(_scurve_prev_leg, _scurve_next_leg, wp_radius, _pos_control.get_lat_accel_max(), _fast_waypoint, _track_scalar_dt * dt, target_pos_3d_ftype, target_vel, target_accel);

It seems that there is a problem with the returned finished。

xianglunkai, 2023-02-17:
In addition, I see that the control tracking is very good.

xianglunkai, 2023-02-17:
I recently found that when setting the PSC parameter, the value of I should be very small or zero, otherwise the turn tracking is not good, which is inconsistent with the setting of 0.2P in the wiki?

rmackay9, 2023-02-20:
Hi @xianglunkai,

What issue are you seeing with the s_finished?

xianglunkai, 2023-02-22:
@rmackay9 
Whether the s_finished return true is delayed some time, resulting in overshoot?