ArduPilot/ardupilot #22958: ESC FC passthrough doesn't work on Matek H743 Wing v3

lotsofone, 2023-02-17:
## Bug report

**Issue details**

I could not get https://esc-configurator.com/ connect to the fc.
I can do it on Matek F405-WMN, but not on Matek H743 Wing v3.
I tried opening an issue here https://github.com/stylesuxx/esc-configurator/issues/327, and asking a question here https://discuss.ardupilot.org/t/esc-configurator-cannot-connect-to-matek-h743-wing/96861

**Version**
Plane 4.3.3

**Platform**
Plane

**Airframe type**
glider

**Hardware type**
Matek H743 Wing v3

**Logs**
https://github.com/stylesuxx/esc-configurator/issues/327

<details>
  <summary>Full parameter list</summary>

  ```txt

  ACRO_LOCKING,0
  ACRO_PITCH_RATE,180
  ACRO_ROLL_RATE,180
  ACRO_YAW_RATE,0
  ADSB_TYPE,0
  AFS_ENABLE,0
  AHRS_COMP_BETA,0.1
  AHRS_EKF_TYPE,3
  AHRS_GPS_GAIN,1
  AHRS_GPS_MINSATS,6
  AHRS_GPS_USE,2
  AHRS_ORIENTATION,4
  AHRS_RP_P,0.2
  AHRS_TRIM_X,0.01095832
  AHRS_TRIM_Y,0.06519349
  AHRS_TRIM_Z,0
  AHRS_WIND_MAX,0
  AHRS_YAW_P,0.2
  ALT_HOLD_FBWCM,0
  ALT_HOLD_RTL,12000
  ALT_OFFSET,0
  ARMING_ACCTHRESH,0.75
  ARMING_CHECK,1
  ARMING_MIS_ITEMS,0
  ARMING_OPTIONS,0
  ARMING_REQUIRE,1
  ARMING_RUDDER,2
  ARSPD_AUTOCAL,1
  ARSPD_BUS,0
  ARSPD_DEVID,0
  ARSPD_FBW_MAX,19
  ARSPD_FBW_MIN,8
  ARSPD_OFFSET,-205.7585
  ARSPD_OPTIONS,11
  ARSPD_PIN,4
  ARSPD_PRIMARY,0
  ARSPD_PSI_RANGE,1
  ARSPD_RATIO,1.818737
  ARSPD_SKIP_CAL,0
  ARSPD_TUBE_ORDER,2
  ARSPD_TYPE,15
  ARSPD_USE,1
[... 1139 lines cut ...]
  TRIM_PITCH_CD,0
  TRIM_THROTTLE,60
  TUNE_CHAN,0
  TUNE_CHAN_MAX,2000
  TUNE_CHAN_MIN,1000
  TUNE_ERR_THRESH,0.15
  TUNE_MODE_REVERT,1
  TUNE_PARAM,0
  TUNE_RANGE,2
  TUNE_SELECTOR,0
  USE_REV_THRUST,2
  VISO_TYPE,0
  VTX_ENABLE,0
  WP_LOITER_RAD,120
  WP_MAX_RADIUS,0
  WP_RADIUS,90
  YAW_RATE_ENABLE,0
  YAW2SRV_DAMP,0
  YAW2SRV_IMAX,1500
  YAW2SRV_INT,0
  YAW2SRV_RLL,1
  YAW2SRV_SLIP,0

  ```
</details>

amilcarlucas, 2023-12-06:
I have tested this on 4.5.0-Dev version and could connect to BLHelliSuite32 . Can you update to at least ArduCopter 4.4.3 and re-test?

andyp1per, 2023-12-06:
This was fixed in 4.4

closed 2023-12-06 by andyp1per