ArduPilot/ardupilot #22960: Xplane crash data inet greather than max!

MdAMALLAL, 2023-02-17:
issue in running SITL with Xplane

**Issue details**
Xplane crash and show
"""
BE ADVISED 
You requested a data inet output value that is greater than max! 
(inet_parse_public.cpp:386)
"""
Arduplane.exe show:

Setting SIM_SPEEDUP=1.000000
Waiting for XPlane data on UDP port 49001 and sending to port 49000
Loaded 20 DRefs from @ROMFS/models/xplane_plane.json
Home: 33.663711 -7.228875 alt=164.480000m hdg=0.000000
Starting sketch 'ArduPlane'
Starting SITL input
Using Irlock at port : 9005
bind port 5760 for 0
Serial port 0 on TCP port 5760
bind port 5762 for 2
Serial port 2 on TCP port 5762
bind port 5763 for 3
Serial port 3 on TCP port 5763
validate_structures:489: Validating structures
New connection on serial port 0
Connected to 127.0.0.1:49000
De-selecting code 8
De-selecting code 25
De-selecting code 29
De-selecting code 58
Selecting 1 data types
X-Plane home reset dist=3825742.974561 alt=0.0/164.5
the stable version do not select nor de-select codes

**Version**
dev

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Logs**
_Please provide a link to any relevant logs that show the issue_

magicrub, 2023-02-17:
What version of ArduPilot? What version of xplane?

MdAMALLAL, 2023-02-17:
Ardupilot master branch,
X-Plane 11

Yu-hub-0, 2024-07-28:
Pls has this problem solved for yours? How is it solved?Is it because of the different versions? I would appreciate it if you could reply to me.