ArduPilot/ardupilot #23006: Plane: Failing takeoff from moving ship

strg-v, 2023-02-22:
We (www.searchwing.org) recently failed to launch a fixed wing UAV from a moving ship. We previously performed test flights in Berlin to ensure the functionality of the vehicle, packed it and shipped it to the Canary Islands. There we unpacked it and tried to launch from a ship. Waves were quite strong, so launching from a standing ship was in this situation not feasible. 

In order to reduce rolling of the ship, we had to start from the moving ship. We launched the UAV via hand start. Movements and directions were like this:

Wind direction -> | <- UAV start direction | Ship movement -> 

Wind was quite strong, so airspeed should have been enough. We do not use an airspeed sensor.

We documented some theories in our wiki: 

https://wiki.searchwing.org/en/home/Development/Flightdocumentation/205

A short summary of a theory we have:

- Log says UAV is already flying, `STAT isFlying` is almost instantly `True` after arming. 
[image]
- Firmware checks ground speed against 1.5m/s to indicate the `isFlying` status: https://github.com/ArduPilot/ardupilot/blob/9c726b8f6896348f0f5c8e3425e12db8a64f372e/ArduPlane/is_flying.cpp#L22
- Flights from ship SeaWatch4 in 2021 showed with a max. ship speed of 1 m/s a different behaviour.
[image]
- Could be that the direction of ground movement is negative in relation to the start direction, even though airspeed is positive, therefore confusing the movement estimation of the vehicle.
- Could be that drone movement estimation is going crazy if `isFlying` is already True.

Do you have any other idea, or can you confirm our theory why our takeoffs failed? We will retry at the start of March, then we can test certain scenarios and collect data for certain scenarios. 

**Version**
4.2.1

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
fixed-wing

**Hardware type**
Pixracer

**Logs**
[Flight Logs + parameters](http://breakout.hs-augsburg.de/searchwing-gui/2023-01-23-17-27-02-maggy-Canaries-FailedTakeoffs/)

booo, 2023-03-26:
This was rather a problem with our configuration than a problem with ArduPilot software.

We autotuned the plane once and changed the wing servos/setup afterwards. In addition one of our colleagues changed the pitch_trim parameter.

@strg-v please close this ticket.

magicrub, 2023-03-27:
Please use an air sensor next time

closed 2023-03-27 by magicrub

booo, 2023-03-27:
Are there waterproof air sensors?