ArduPilot/ardupilot #23042: AP_Mount: add MNTx_SYSID param to easier support following another vehicle

rmackay9, 2023-02-27:
Similar to Copter's follow mode, AP_Mount should have a SYSID parameter that allows users to more easily specify what vehicle they would like the gimbal to point at when in ROI_SYSID mode.

arikrupnik, 2023-02-27:
This should probably also add `5: SysId Target` as a valid value to [MNTx_DEFLT_MODE](https://ardupilot.org/copter/docs/parameters.html#mnt1-deflt-mode-mount-default-operating-mode)

khanasif786, 2023-03-30:
@rmackay9 I am working on it.

khanasif786, 2023-03-31:
@rmackay9 Adding a new param will introduce an ambiguity when **MAV_CMD_DO_SET_ROI_SYSID** is sent and then again parameter is changed in the same flight.

rmackay9, 2023-03-31:
@khanasif786,

Re the first question, yes, I totally agree we just need to update the MNTx_DEFLT_MODE parameter description and make sure (in SITL) that AP_Mount really changes to that mode soon after startup.

Re the 2nd question, I think if we have a new parameter it will just be used as the default until a do-set-roi-sysid command arrives.  Users may not have an easy way to send a do-set-roi-sysid so having this param would allow them to set it easily... so really it might be better named something like MNTx_SYSID_DEFAULT.

PR #23377 (2023-04-24): AP_Mount: add MNTx_SYSID_DFLT param to easier support following another vehicle
Fixes #23042
closed 2023-04-24 by rmackay9