ArduPilot/ardupilot #23046: KakuteH7v2 no buzzer

freddybloggs, 2023-02-27:
## Bug report

**Issue details**

_Please describe the problem_
I have a kakuteH7V2 and the buzzer is not working, I have tried every setting under the sun and failed to get a peep out of it. I programmed the FC with inav just to make sure it wasn't the board or my wiring. I can confirm the buzzer works with inav using PC13.

**Version**
4.40dev

_What version was the issue encountered with_
4.40dev
**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
QuadX
**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
KakuteH7v2
**Logs**
_Please provide a link to any relevant logs that show the issue_

andyp1per, 2023-02-27:
How are you testing? You will only get a single tone on arming and that's about it. If you want more complicated tones you need to use one of the spare motor outputs as an alarm.

freddybloggs, 2023-02-27:
I was expecting an initialisation tone as my omnibusf7 ( even with it's buzzer pin set to -1 ) makes a feeble attempt at a tone when it starts up, I wasn't arming it because I didn't want the props blowing bits all over my desk !!!.
I have since taken it outside and now i get an arming beep sorry to have bothered you.

closed 2023-02-27 by andyp1per