ArduPilot/ardupilot #23053: AP_Mount: combination of angle + angle_rate not possible with MAV_CMD_DO_GIMBAL_MANAGER_PITCHYAW

bengterik, 2023-02-27:
## Bug report

**Issue details**

I am controlling a servo-gimbal over UDP using mainly the MAV_CMD_DO_GIMBAL_MANAGER_PITCHYAW (# 1000) message, and according to the MAVLink docs one is to be able provide a combination of angles and angle rates, but currently only one or the other is possible at once.

To exemplify, the mavproxy command:

    message COMMAND_LONG 0 0 1000 0 20 10 5 5 0 0 0

should move the gimbal to (20,10) with 5 rad/s in each axis but instead it moves there as fast as it can, resulting in the same behaviour as if one would send:

    message COMMAND_LONG 0 0 1000 0 20 10 float("NaN") float("NaN") 0 0 0

Providing only angle rates works, but only if one of the angles, i.e. param 3 and 4, are set to NaN. 

For example:

    message COMMAND_LONG 0 0 1000 0 float("NaN") float("NaN") 5 5 0 0 0

makes the gimbal move at 5 degs/s until it's stopped by another command.

I would be happy to contribute and I have an idea about what might be causing the issue which I will post below - thankful for any thoughts or tips!

**Version**
ArduPlane 4.3.3

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Flying wing

**Hardware type**
Pixracer r15, using servo-gimbal

bengterik, 2023-02-27:
From my understanding it is due to the implementation in AP_Mount.cpp where the message handler returns immediately after setting the new target angles, disregarding any rates provided. 

In the function handle_command_do_gimbal_manager_pitchyaw(const mavlink_command_long_t &packet):

    // param1 : pitch_angle (in degrees)
    // param2 : yaw angle (in degrees)
    const float pitch_angle_deg = packet.param1;
    const float yaw_angle_deg = packet.param2;
    if (!isnan(pitch_angle_deg) && !isnan(yaw_angle_deg)) {
        set_angle_target(gimbal_instance, 0, pitch_angle_deg, yaw_angle_deg, flags & GIMBAL_MANAGER_FLAGS_YAW_LOCK);
        return MAV_RESULT_ACCEPTED;
    }

    // param3 : pitch_rate (in deg/s)
    // param4 : yaw rate (in deg/s)
    const float pitch_rate_degs = packet.param3;
    const float yaw_rate_degs = packet.param4;
    if (!isnan(pitch_rate_degs) && !isnan(yaw_rate_degs)) {
        set_rate_target(gimbal_instance, 0, pitch_rate_degs, yaw_rate_degs, flags & GIMBAL_MANAGER_FLAGS_YAW_LOCK);
        return MAV_RESULT_ACCEPTED;
    }

**Potential solution**

My first idea for a solution was to switch the order of the two if-statements, removing the return statement inside the one setting the rates, but after some further investigation I’m guessing a third MountTargetType with some condition (checking if the angle has been reached) also has to be introduced in the following code from AP_Mount_Servo::update()

    case MAV_MOUNT_MODE_MAVLINK_TARGETING: {
            switch (mavt_target.target_type) {
            case MountTargetType::ANGLE:
                _angle_rad = mavt_target.angle_rad;
                break;
            case MountTargetType::RATE:
                update_angle_target_from_rate(mavt_target.rate_rads, _angle_rad);
                break;
            }
            // update _angle_bf_output_deg based on angle target
            update_angle_outputs(_angle_rad);
            break;
        }

...although this would involve modifications to each of the gimbal types which felt a bit daunting being my first PR... 

As said I'm thankful for any input/thoughts on this!

rmackay9, 2023-02-28:
.. thanks for raising this issue.  One underlying problem is though that almost none of the gimbals themselves support using both angle and rate controls at the same time.

Also it is important to clarify when rate is used, whether this is the target rate or the maximum rate.  In this case I think you're requesting that this be interpreted as the maximum rate that the gimbal will move but this is not how we normally interpret a message with both target position and rate (at least when we're talking about controlling the vehicle).

bengterik, 2023-03-01:
Thanks for your reply!

For our use case, we thought setting the rate would be nice to reduce the jerkiness of the camera when doing larger angle steps, both in MAVLINK_TARGETTING-mode but also in other modes such as GPS_POINT where changing points sometimes becomes disorienting due to the speed of the servos.

Going back to the PITCHYAW command, if I'm understanding you correctly, it is too early to allow for both angles and turn rates with the current lack of gimbal support?

rmackay9, 2023-03-01:
@bengterik,

Which kind of gimbal are you using?  Is it a 3-axis servo gimbal?  e.g. actual servos?  In this case, what we really need is some new parameters to allow specifying the maximum speed that the servos should move at.

bengterik, 2023-03-01:
Correct, it is a custom construction with three SG90 servos that we have mapped with SERVOx_FUNCTION to Mount-Tilt/Roll/Pan.

I came across the slew functions in the servo code as well as the parameters setting the slew rate for other servo functions (thrust, pitch/yaw for the aircraft). I guess this could be implemented for the MountTilt/Roll/Pan-functions too, or just the SERVOx itself to make it accessible for other servo applications as well. What do you think?