ArduPilot/ardupilot #23064: Arducopter_Simulink: Conversion of thstOut from Motor Mixer subsystem block to PWM values

MarcLeclercGit, 2023-02-28:
**Feature that I would like:**
Hi,
I started using the the Ardupilot simulink simulation recently for a research project due to its high fidelity to the real architecture and I would like to make the conversion of the motor mixer output of each motor (thstOut) to PWMs values. Our experimental  drone is an octoquad X8 and I made a MATLAB script that calculate the 6dof dynamics of the drone and the changing thrust and pitching moments of the propellers function of the airspeed.  I use this approach instead of using the lumped mixer-motor-drone in a transfer function to better observe the influence of different parameters on the drone's behavior.

The coaxial propellers thrust/moments function is done with a neural net using multiple dataset from wind tunnel tests. The coaxial propellers influence each other so it was simpler to use experimental data instead of a complex analytical model. For each arm, this neural net function takes in the PWM values from the upper and lower motors and the incoming airspeed and then output the total thrust and moments in all axis for that arm.

**What I tried**
To convert the value of each motor throttle to PWMs values, I used the simple relation for each 8 motors 
PWM(i,:) = thstOut(i,:)*PWMrange + PWMmin on the thstOut variable.  To validate it, I inputted a flight log done in stabilize and compared the PWMs of the Simulink Motor Mixer to those from the flight logs. The simulated and real PWM curves a very similar in shape(see Graph_compare_PWM_SimVsExp), but there're offsets in the values for most of the flight. I highly suspect that my simple PWM conversion is not what is used by Ardupilot and caused these variations.
[image]

I looked in the ardupilot repository and  SRV_Channels.cpp seems to apply a slewrate and final scaling before converting it to a PWM, but I am not coding in C++ often so I'm not quite sure to how it done and if it's the function that I'm looking for.

**Describe the solution you'd like**
A conversion from thstOut variable which correspond to the individual 8 motor throttle (0-1) to PWMs values that match the conversion done in Ardupilot.

**Note**
In the getMotorMixerFactors.m I added a  new case for an Octoquad X8 in the in the Octa and OctaQuad since it was not available.
I used :
% Coax X8
  axis_facs_motors_pre = { [45, 1]; [-45, -1]; [-135, 1]; [135, -1]; ...
                        [-45, 1]; [45, -1]; [135, 1]; [-135, -1] };

**Version**
Master branch

**Additional context**
 @fbredeme @peterbarker have you ever looked to convert the thstOut  variable to PWM values?
 Thanks

IamPete1, 2023-03-01:
Motors does it own conversion here:
https://github.com/ArduPilot/ardupilot/blob/a6041f77e8f282c763b46f0718b31c8fb0561a7f/libraries/AP_Motors/AP_MotorsMulticopter.cpp#L470-L475

And:

https://github.com/ArduPilot/ardupilot/blob/a6041f77e8f282c763b46f0718b31c8fb0561a7f/libraries/AP_Motors/AP_MotorsMulticopter.cpp#L451-L468

And: 

https://github.com/ArduPilot/ardupilot/blob/a6041f77e8f282c763b46f0718b31c8fb0561a7f/libraries/AP_Motors/AP_MotorsMulticopter.cpp#L355-L370

Called from here:
https://github.com/ArduPilot/ardupilot/blob/a6041f77e8f282c763b46f0718b31c8fb0561a7f/libraries/AP_Motors/AP_MotorsMatrix.cpp#L171

Issue is not really the right place for such questions, but feel free to jump on our discord, we have simulation and research channels. https://discord.gg/c684xXhg

closed 2023-03-01 by IamPete1

NguyenBinh3112, 2024-06-03:
Hello 2 guys, i'm finding a way to convert thrust to PWM signal. I've read in ardupilot but i do not understand algorithm. Can you send me document, which you use to convert thrust to PWM signal in AP_MotorsMulticopter.c, pls ?