ArduPilot/ardupilot #23117: Copter: Is there any problem in function euler_accel_limit() ? version-2

luweiagi, 2023-03-07:
I think there is a bug in this function:
[AC_AttitudeControl::euler_accel_limit(const Vector3f &euler_rad, const Vector3f &euler_accel)](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_AttitudeControl/AC_AttitudeControl.cpp#L908)

lthall, 2023-03-24:
I believe this is correct. I am sorry it took so long to get to this. It has been on my lists of things to do!!

lthall, 2023-03-24:
#23321

luweiagi, 2023-03-24:
> I believe this is correct. I am sorry it took so long to get to this. It has been on my lists of things to do!!

> I believe this is correct. I am sorry it took so long to get to this. It has been on my lists of things to do!!

Thanks for your reply. These years I always remember this bug/problem. Now my suggestion is accepted, I feel happy although this is a very tiny problem.

I think fix this bug can enhance copter maneuverability by releasing the limit.

luweiagi, 2023-03-24:
> I believe this is correct. I am sorry it took so long to get to this. It has been on my lists of things to do!!

Thanks for your reply. I am happy for being accepted. These years this problem/bug is always in my head. Although it is a tiny bug with little affect on maneuverability, but let me be happy for a while.

luweiagi, 2023-08-23:
I reply https://github.com/ArduPilot/ardupilot/pull/23326 at here:

rmackay9, 2024-01-24:
Can we close this now that https://github.com/ArduPilot/ardupilot/pull/23326 has been merged?

closed 2024-01-24 by rmackay9