ArduPilot/ardupilot #23128: Feature request: Add heading flight mode

omred173, 2023-03-07:
Hi everyone, First time I write here
I’ll like to offer flight mode that i use in big uav for none-gps flight at BVLOS ( as a manual dead reckoning), And also required when you fly BVLOS with ATC ( air traffic control)
Heading flight : allowing to fly specific magnetic headings by enter 3 digit in the GCS, The airspeed / ground speed for multirotor, and altitude will be controlled as FBWB or cruise mode, but as i know cruise/ FBW-B mode fly ground track, i case of gps loss its use dead reckonings including last known wind to fly and without good sensor it will drift very fast.

we need to fly magnetic heading as it’s reading by the compass.
It’s also the standard when ATC ask you to fly heading

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

Rolf-G, 2023-03-29:
Hi Omred173,

just my 2 cents for Arduplane: 

To fly BVLOS according to the magnetic heading of the ATC instructions, it will be easiest to write a small LUA script that first calculates the magnetic heading and sends it to the ground station so that you know what to answer when you are asked for the heading.  With a LUA script you could also do the opposite, so that Arduplane in FBWB strictly adheres to the preselected magnetic heading. 

In which mode do you fly Arduplane BVLOS so far? Actually, it's no big deal to convert the GPS truecourse to magnetic course. Depending on the crosswind component, you add a lead angle and have the magnetic heading. A LUA script could take care of this if you don't have an experienced full scale pilot at your side.

Rolf

omred173, 2023-04-01:
hi Rolf,
Thanks for your response

when we fly BVLOS we use guided mode/ AUTO.
The idea is also not to use gps, because blockages are quite common

timtuxworth, 2025-07-01:
This already exists. GUIDED mode can be set to fly to a heading at an altitude. Unfortunately (currently) no ground station software supports this, so you can only enter it via Lua script or by sending mavlink messages. You can do this by sending COMMAND_INT mavlink messages in MavProxy.py or using pymavlink.
Maybe you could encourage QGC or Mission Planner devs to add a "fly to heading" UI.