ArduPilot/ardupilot #23149: AP_Mount: Siyi gimbals should use angle control where possible

rmackay9, 2023-03-09:
The existing [AP_Mount_Siyi gimbal driver](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount_Siyi.h) ([used to control the ZR10, ZR30 and A8 gimbals](https://ardupilot.org/copter/docs/common-siyi-zr10-gimbal.html#common-siyi-zr10-gimbal)) uses the gimbal's rate control API to move the gimbal.  This means the driver must use an internal P controller to convert the desired angle to a desired rate based on the latest reported angles from the gimbal.

Since the driver was first developed, SIYI has added an angle control API so we should update the driver to use this and remove the P controller.

khanasif786, 2023-05-28:
@rmackay9 Assign this to me please.

rmackay9, 2023-05-28:
Hi @khanasif786, great, thanks.

I've already done some work on it here: https://github.com/rmackay9/rmackay9-ardupilot/tree/mount-siyi-angle-control

but I found the gimbal yaw didn't move as expected.  It would momentarily stop each time it received a new attitude command.