ArduPilot/ardupilot #23154: AI & ArduPilot development environment

rmackay9, 2023-03-09:
There are lots of ways to use AI with drone, too many to cover in this single enhancement request but I think a good start would be making it easier for developers to test externally developed models with ArduPilot vehicles in SITL.

This enhancement request involves creating scripts and wiki documentation to help developers setup a development and testing environment for vision based AI models to be integrated with ArduPilot vehicles (primarily multicopters, rovers and boats).

- [ ] decide which AI models will be supported (perhaps [TensorFlow ](https://www.tensorflow.org) and/or [OpenCV AI](https://opencv.org)
- [ ] choose a simulator (probably [webots](https://ardupilot.org/dev/docs/sitl-with-webots.html) or [AirSim](https://ardupilot.org/dev/docs/sitl-with-airsim.html))
- [ ] ensure the simulator's onboard camera output is available to the AI model
- [ ] ensure the AI model's outputs can control the vehicle
- [ ] create scripts to setup the development environment
- [ ] create documentation on how to setup and use the development environment
- [ ] create a ready-to-use example demonstrating the use of the environment.  Ideas include
  - [ ] multicopter safe landing (see [Open Lander](https://github.com/stephansturges/OpenLander))
  - [ ] multicopter wind turbine inspection (turn the gimbal to face the turbine blade)
  - [ ] rover lane following

yalexx, 2023-03-22:
How can i contribute ?

akileshas, 2025-01-10:
@rmackay9 can u just elaborate the actually objective of this issue and if the issue is under my scope of field i can take this issue and proceed with it

rmackay9, 2025-01-10:
Hi @akileshas,

I think looking at what Rhys, Asif Khan and Sanket Sharma did in the 2024 GSoC projects would be best.  In particular Asif and Rhys used Gazebo for testing.  We should make sure that setup is easy to setup