ArduPilot/ardupilot #23156: Change altitude during missions

highfreq, 2023-03-09:
I have been looking forward to the possibility of changing altitude during a mission for years now. It is always, maybe, in the next release.

IamPete1, 2023-03-09:
Were going to need some more information here. Change the altitude how? What vehicle type? You can put in waypoints with different altitudes and upload updated waypoints as the mission is running.

highfreq, 2023-03-09:
Lets say i plan a mission at 30 meters (copter). Than during the mission i want it to be at 35, there is no way of achieving it. Been asking this for years, at the moment i use a  self modified firmware to do it.

IamPete1, 2023-03-09:
You can upload updated waypoints during the mission. If you have a working branch you could open a PR for consideration.

highfreq, 2023-03-09:
I need to change altitude 10 times a minute, i'll stick with my modified firmware. Tried 2 years ago and someone told me it was too much of a hack, have 1500 flights on the hack, never a problem.

highfreq, 2023-03-09:
Anyway Pete, you shouldn't talk to me, i am the problematic user banned from ardupilot discuss because i argued with SIYI.

IamPete1, 2023-03-09:
I'm still not really sure I understand the use case, how do you want to modify the altitude? From a GCS? From the transmitter? Why do you need to change the altitude? Could rangefinder and the surface tracking functionality be used? 

The problem with "hacks" is not that the don't work, they work great. It is that they break things for other uses and/or make development or maintenance of other features hard.

rmackay9, 2023-03-09:
This issue is already on the to-do list so I'm going to close this issue if that's OK.

- https://github.com/ArduPilot/ardupilot/issues/11251
- https://github.com/ArduPilot/ardupilot/issues/16177

closed 2023-03-09 by rmackay9

highfreq, 2023-03-09:
Has been on the to do list for years. Fuck you Randy

graheeth, 2023-10-24:
I changed the firmware slightly to allow the throttle input to be able the control the altitude height of the waypoints so there is a way to let you control altitude during automode. heres the link:
https://github.com/graheeth/AutoModeAltControl.git