ArduPilot/ardupilot #23180: Landing sequence flag not cleared on first RC connect when disarmed

Hwurzburg, 2023-03-11:
Arming is prevented when powering the autopilot without RC (goes to FS RTL) and there is a DO_LAND_START so the autopilot sets the in landing seq flag..when RC is connected and switched out of RTL (to ACRO), the flag is NOT cleared resulting in a PREARM failure until the mode is switched to one that entry clears the flag and arming is then allowed
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

tridge, 2023-03-11:
that is not actually a bug, it is there because of a real issue that happens like this:
 - user switches to QLOITER or QHOVER to arm (common with quadplanes so you can see motors spinning)
 - user then switches to AUTO to start the mission

we could add something that resets the mission on first R/C connect if the plane is disarmed.

Hwurzburg, 2023-03-11:
sorry that case does not make sense to me....arming in QLOITER means that the landing sequence flag is already cleared, or it would not arm....that is not my issue....without RC the RTL is entered...if you have DO_LAND_START as the RTL action, its set the in landing sequnce flag....when RC connects and changes it to say ACRO, the flag is not cleared and prearm is in effect

tridge, 2023-03-11:
it would fix it as the mission reset on first R/C connect would clear the flag as a by-product. Clearing the flag without a mission reset would be incorrect as AUTO is still in the landing sequence.
The point of the arming check is to ensure that after arming when AUTO is entered that we don't start a landing sequence

Hwurzburg, 2023-03-12:
one gotcha to using the mission reset....if DO_LAND_START is the first item, it immediately re-establishes the in landing sequence after a mission.reset()...it works if there is a non landing mission item for item 1

timtuxworth, 2023-03-12:
> we could add something that resets the mission on first R/C connect if the plane is disarmed.

This actually sounds like a great idea. If the "first R/C connect" also includes power up, then it also fixes the other lingering issue about getting stuck in landing sequence if you manually take over to land a plane after an RTL or mission landing sequence. Of course the key is _if the plane is disarmed_.

peterbarker, 2024-05-07:
@Hwurzburg Did @IamPete1 's RC-failsafe PR resolve this issue?

Hwurzburg, 2024-05-12:
I think it does

closed 2024-05-12 by Hwurzburg