ArduPilot/ardupilot #23181: Rudder arming allowed in ALL modes

Hwurzburg, 2023-03-11:
Originally (and the param still has it in its description) rudder arming could only occur in pilot controlled modes....currently rudder arming is allowed in all modes....this seems a safety issue, especially for AUTO  and TAKEOFF when you may have a full throttle takeoff as the first item, hard rudder would start it moving rapidly to the pits or into people...I was under the impression that other autothrottle modes (RTL,CRUISE,LOITER) use the is_flying check to prevent arming, but have not checked in SITL....

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

IamPete1, 2023-03-12:
I think this one is going to have to be a docs update. We can't just stop people arming in a way that currently works, we will be snowed under with issue reports (happy to add a option that defaults off tho).

For the most part if you arm in a auto throttle mode the throttle will be suppressed until you raise the throttle. 

I presume you had plane in mind here? Of course you can arm in auto throttle modes in copter too....

Hwurzburg, 2023-03-12:
I disagree....its a safety issue....and easily fixed....see #23186

rmackay9, 2024-01-22:
@Hwurzburg I guess we can close this one now?