ArduPilot/ardupilot #23192: Copter/Rover-4.4.0 issues list

rmackay9, 2023-03-13:
This is a list of issues discovered during Copter and Rover 4.4 testing.  The older [4.3 issues list can be found here](https://github.com/ArduPilot/ardupilot/issues/21711).

Reports requiring investigation:

- [ ] Terrain alt clamping can lead to failed takeoff and flip on takeoff in Auto (reported by rmackay9)

Confirmed bugs:

- none

oraziotorre, 2023-03-16:
Regarding "RTL_SPEED does not work if WPNAV_SPEED adjusted (reported by lthall)", I have noticed that when starting a WP mission and subsequently switching to the RTL mode, the RTL_SPEED parameter is no longer considered and the WPNAV_SPEED parameter is used instead. Is this intentional? Can I open a pull request to improve this issue?

rmackay9, 2023-03-16:
@oraziotorre, normally when the vehicle switches into RTL it should use the RTL_SPEED parameter unless it is zero in which case it should fall back to WPNAV_SPEED.  It that's not what it's doing then it sounds like a problem.

hyperion11, 2023-05-13:
[image]
Flywoo GOKU GN 745 AIO, custom firmware with tcal enabled (disable some features). IMU calibration successful. Baro not starting.