ArduPilot/ardupilot #23193: Benchmark build failure on g++11 Ubuntu 22.04

Ryanf55, 2023-03-13:
## Bug report

**Issue details**

When trying to run `Tools/scripts.build_ci.sh` locally, it has an error compiling gbenchmark. 

**Version**
Latest master
```bash

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

Ryanf55, 2023-03-13:
This has been fixed upstream already. Can we update gbenchmark? 
https://github.com/google/benchmark/commit/3d1c2677686718d906f28c1d4da001c42666e6d2
https://github.com/google/benchmark/pull/1060

khancyr, 2023-03-13:
yes, we need to update to the latest release. It isn't blocking the rest so that is safe

closed 2023-03-14 by Ryanf55