ArduPilot/ardupilot #23202: CI not catching build errrors on GCC11 (Ubuntu 22.04

Ryanf55, 2023-03-14:
## Bug report

As found in #23193, I ran into an issue of lack of compilation on Ubuntu 22.04. Considering some/many developers are using 22.04, it would be good for CI to also check this works. Could we upgrade some of the CI runners to check things are stable in 22.04? 

Need to consider what minimum [Ubuntu version ](https://wiki.ubuntu.com/Releases) we should check in CI. Considering 18.04 is almost deprecated, perhaps CI should just check 20.04 and 22.04. 

**Version**
`master` hash c651a82196c129012f31ad0e0886829c1e8a0a0a

khancyr, 2023-03-14:
CI is running on 20.04 that is matching our release server. 
18.04 is to forgot as python2 support is a mess.
We should add 22.04 to the environnement setup workflow, that is a CI run that test our install script and run a build once a week.

Otherwise we need a team decision to move up everything to 22.04.

Ryanf55, 2023-03-16:
If we add the ROS2 build using the same CI infrastructure, since we are targeting `humble` on 22.04, it would help resolve this.

magicrub, 2023-03-16:
I use Ubuntu 22 on WSl2 and I'm in favor of moving to 22.04 as well

Ryanf55, 2023-03-22:
18.04 is officially EOL on April 30. Since 4.5 will release after that, perhaps this is a good time to deprecate 18.04. Then, CI just needs to check support on on 20.04 and 22.04.

closed 2023-03-23 by peterbarker