ArduPilot/ardupilot #23207: Simulation on Hardware

hassankord, 2023-03-14:
## Bug report

**Issue details**

running 
./waf configure \
  --board=Pixhawk1-1M \
  --default-param="plane-default.param" \
  --extra-hwdef="plane-extra-hwdef-sitl-on-hw.dat" \

./waf plane

[ 58/954] Compiling libraries/AP_Airspeed/AP_Airspeed_Health.cpp
In file included from ../../libraries/AP_AHRS/AP_AHRS_SIM.h:28,
                 from ../../libraries/AP_AHRS/AP_AHRS_SIM.cpp:1:
../../libraries/AP_AHRS/AP_AHRS_Backend.h:114:31: error: 'Location' has not been declared
  114 |     virtual bool get_location(Location &loc) const WARN_IF_UNUSED = 0;
      |                               ^~~~~~~~
compilation terminated due to -Wfatal-errors.

Waf: Leaving directory `/cygdrive/c/ardupilot/build/Pixhawk1-1M'
Build failed
 -> task in 'objs/AP_AHRS' failed (exit status 1):
        {task 123145285459024: cxx AP_AHRS_SIM.cpp -> AP_AHRS_SIM.cpp.0.o}
 (run with -v to display more information)

**Version**
Release 4.3.4 1st March  2023

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
flying wing

**Hardware type**
Pixhawk1-1M

**Logs**
++ ./waf configure --board=Pixhawk1-1M --default-param=plane-default.param --extra-hwdef=plane-extra-hwdef-sitl-on-hw.dat
Setting top to                           : /cygdrive/c/ardupilot
Setting out to                           : /cygdrive/c/ardupilot/build
Autoconfiguration                        : enabled
Setting board to                         : Pixhawk1-1M
Using toolchain                          : arm-none-eabi
Checking for 'g++' (C++ compiler)        : /cygdrive/c/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-g++
Checking for 'gcc' (C compiler)          : /cygdrive/c/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-gcc
Checking for c flags '-MMD'              : yes
Checking for cxx flags '-MMD'            : yes
CXX Compiler                             : g++ 10.2.1
Checking for program 'make'              : /usr/bin/make
Checking for program 'arm-none-eabi-objcopy' : /cygdrive/c/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-objcopy
Default parameters                           : plane-default.param
Including /cygdrive/c/ardupilot/libraries/AP_HAL_ChibiOS/hwdef/Pixhawk1/hwdef.dat
Including /cygdrive/c/ardupilot/libraries/AP_HAL_ChibiOS/hwdef/fmuv3/hwdef.dat
Removing AP_BATTERY_SMBUS_SOLO_ENABLED
Removing BOARD_CHECK_F427_USE_1M
Including /cygdrive/c/ardupilot/libraries/AP_HAL_ChibiOS/hwdef/include/minimal_GPS.inc
Including /cygdrive/c/ardupilot/libraries/AP_HAL_ChibiOS/hwdef/include/minimize_features.inc
Removing STORAGE_FLASH_PAGE
Adding environment SIM_ENABLED 1
Setup for MCU STM32F427xx
Writing hwdef setup in /cygdrive/c/ardupilot/build/Pixhawk1-1M/hwdef.h
Writing DMA map
Setting up as normal firmware
No change in hwdef.h
Generating ldscript.ld
No default parameter file found
Checking for env.py
env set SIM_ENABLED=1
env set ENABLE_DFU_BOOT=0
env set WITH_FATFS=1
env set PROCESS_STACK=0x1C00
env set MAIN_STACK=0x600
env set IOMCU_FW=0
env set PERIPH_FW=0
env set HAL_NUM_CAN_IFACES=2
env set HAL_CANFD_SUPPORTED=0
env set BOARD_FLASH_SIZE=1024
env set EXT_FLASH_SIZE_MB=0
env set ENABLE_CRASHDUMP=0
env set APP_RAM_START=None
env set CPU_FLAGS=['-mcpu=cortex-m4', '-mfpu=fpv4-sp-d16', '-mfloat-abi=hard', '-DARM_MATH_CM4', '-u_printf_float']
env set CORTEX=cortex-m4
env set APJ_BOARD_ID=9
env set APJ_BOARD_TYPE=STM32F427xx
env set USBID=0x1209/0x5741
env set BOOTLOADER_EMBED=1
env set FLASH_RESERVE_START_KB=16
env set EXT_FLASH_RESERVE_START_KB=0
env set FLASH_TOTAL=1032192
env set HAS_EXTERNAL_FLASH_SECTIONS=0
env set CHIBIOS_BUILD_FLAGS=USE_FATFS=yes CHIBIOS_STARTUP_MK=os/common/startup/ARMCMx/compilers/GCC/mk/startup_stm32f4xx.mk CHIBIOS_PLATFORM_MK=os/hal/ports/STM32/STM32F4xx/platform.mk MCU=cortex-m4 ENV_UDEFS=-DCHPRINTF_USE_FLOAT=1
Enabling ChibiOS asserts                     : no
Disabling Watchdog                           : no
Enabling malloc guard                        : no
Enabling ChibiOS thread statistics           : no
Enabling -Werror                             : yes
Checking for intelhex module:                : OK
Enabled OpenDroneID                          : no
Enabled firmware ID checking                 : no
GPS Debug Logging                            : no
Enabled custom controller                    : no
Checking for HAVE_CMATH_ISFINITE             : no
Checking for HAVE_CMATH_ISINF                : no
Checking for HAVE_CMATH_ISNAN                : no
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : no
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : no
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : no
Checking for header endian.h                   : not found
Checking for header byteswap.h                 : not found
Checking for HAVE_MEMRCHR                      : no
Configured VSCode Intellisense:                : no
Checking for program 'python'                  : /usr/bin/python
Checking for python version >= 2.7.0           : 3.7.12
Checking for program 'python'                  : /usr/bin/python
Checking for python version >= 2.7.0           : 3.7.12
DSDL compiler                                  : /cygdrive/c/ardupilot/modules/uavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc
Source is git repository                       : yes
Update submodules                              : yes
Checking for program 'git'                     : /usr/bin/git
Gtest                                          : STM32 boards currently don't support compiling gtest
Checking for program 'arm-none-eabi-size'      : /cygdrive/c/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-size
Benchmarks                                     : disabled
Unit tests                                     : disabled
Scripting                                      : disabled
Scripting runtime checks                       : enabled
Debug build                                    : disabled
Coverage build                                 : disabled
SITL 32-bit build                              : disabled
Checking for program 'rsync'                   : /usr/bin/rsync
'configure' finished successfully (2.061s)

magicrub, 2023-03-14:
@hassankord I see that you're trying to build on an old Pixhawk 1MB hardware. As stated [in the wiki Limitations](https://ardupilot.org/dev/docs/sim-on-hardware.html#limitations) area:

> While it is possible to run on flash-constrained (1MB) boards, many other features will need to be compiled out to allow it to fit.
While the build error you're getting is not necessarily flash-size related, that particular build target was likely never tested.

Unless your hardware is actually very very old, it is likely that you actually have a 2MB version. Try "Pixhawk1".

hassankord, 2023-03-16:
thanks for your reply.
Yes ... the hardware  is very very old.   the code is built on fumv3 without problem(2MB).. but the problem persist when I try fmuv2( 1MB). and it is one of the build option that was not excluded.

peterbarker, 2024-02-19:
@hassankord thanks for the report.  I believe this was fixed with c30bceb2a09a56779c64fee61ea75445831b2b2b

Please re-open if not so.

closed 2024-02-19 by peterbarker