ArduPilot/ardupilot #23219: End of Docker Organisation free plan

khancyr, 2023-03-15:
From April 2023, Docker close the  Docker Team free plan

We were using this to have Docker Org on DockerHub and have our CI Docker Containers https://hub.docker.com/orgs/ardupilot/repositories

We are using Docker on CI to have a few advantage : 

> package versions are fixed
> we pull a fixed image, that is fast and don’t rely on doing apt-get on github cache or ubuntu server
> all toolchain are already installed
> images are reduced to the minimal
> allow the docker image to be used on all CI system

With the closing of the Free Team plan, we have some options: 

> create a new docker user to hold our images
> accept to pay 300$/y (for now, no guarantee it won’t raise), just to have org
> apply to the Docker-sponsored Open source program : this need restructuration (mostly wording to separate ardupilot from the fundation paid service) on our side
> use another docker registry provider like github
> deploy our own registry (please no)
> stop using docker and setup back everything on github action

The main issue on this is that any changes on the CI : changing the docker repository or getting back is setting up github action, will lead to broken CI for old release and surely all partners CI relying on our docker images.

Using back setup script for github action isn’t a hard task, but could lead to some slowdown on CI as we will rely on github apt-cache relay or ubuntu apt server to work. For chibios toolchain we could cache it into github to not have to pull it from our server each time

khancyr, 2023-03-15:
some interesting reading : https://blog.alexellis.io/docker-is-deleting-open-source-images/
Looks like we better get rid of docker for now ... even at the cost of breaking previous CI jobs. We could migrate to github registrery or make back the setup scripts.
It doesn't seem that applying to Docker sponsored Open source program is something we can trust

Vad1mo, 2023-03-15:
There is a [Docker-Sponsored Open Source Program](https://www.docker.com/community/open-source/application/) which might be applicable to you.

If, for some reason, this isn't working out, we are happy to help you out. We operate a Harbor-based [container registry as a service ](https://container-registry.com/)that has many benefits to other registries.

khancyr, 2023-03-17:
Docker accept us on the Sponsored program so we are fine for now !

closed 2023-03-17 by khancyr