ArduPilot/ardupilot #23224: GSF failiures on fixed wing

tridge, 2023-03-15:
The GSF yaw estimator can sometimes produce a very bad yaw. This particularly affects fixed wing aircraft with no compass. EKF3 initialises after launch using GSF and can result in the EKF getting bad roll/pitch estimates.
This issue is to collect examples and work on a fix

Examples:
 - https://discuss.ardupilot.org/t/ekf3-yaw-alignment-on-v4-4-dev-brought-my-plane-down/98076 log http://uav.tridgell.net/GSF-failures/00000071.BIN

priseborough, 2023-03-19:
Log http://uav.tridgell.net/GSF-failures/00000071.BIN has higher than usual levels of GPS velocity noise. I've requested access to the other log. I will try adjusting the GSF use thresholds first. Use of a  GPS velocity vector for yaw can give bad results with low wing loading aircraft that are had launched on wind days as the aircraft can go sideways after launch. 

It's not clear to me how the use of an airspeed sensor to bound yaw error can work without requiring assumptions about launching into wind.

tridge, 2023-03-21:
@priseborough if we waited longer to gain more ground speed for init then using GPS velocity vector for yaw would perhaps be more reasonable?

priseborough, 2023-04-01:
I'm working on a fix for this that will allow the special cases such as planes that don't wish to use the GSF, eg those hand launching without a compass or those that need to do aerobatics to not use the GSF by setting the EK3_GSF_USE_MASK parameter to zero. This is a work in progress and still requires  a patch that will do a sanity check of GPS velocity innovations after the yaw reset to verify that the yaw reset was successful.

Flight testing of the aerobatics scripting has highlighted the limitations of the GSF yaw estimators simplified horizontal motion model.

https://github.com/priseborough/ardupilot/commits/pr-ekf3PlaneYawReset