ArduPilot/ardupilot #23241: AP_Arming: add check that CAN_P1_DRIVER and CAN_P2_DRIVER are not the same

rmackay9, 2023-03-17:
We had a [support issue on the forums](https://discuss.ardupilot.org/t/unable-to-use-gps-and-rangefinder-on-can-ports-1-and-2/97535) where the user was unable to get a DroneCAN rangefinder to work and after some back-and-forth the issue was caused by the user setting both CAN_P1_DRIVER and CAN_P2_DRIVER to "1".  We should add an arming check to ensure these two parameters are never both 1 or both 2.

As a side note, I think the abstraction allowing users to setup CAN_P1_DRIVER to anything but 0 or 1 is an unnecessary complexity.  Why would it be better to have CAN_P1 as the 2nd CAN driver?  We should have just had CAN1_xxx and CAN2_xxx.

peterbarker, 2024-11-23:
Re-using the same driver on both ports conserves resources.

rmackay9, 2024-11-25:
Yes, we can close this now

closed 2024-11-25 by rmackay9