ArduPilot/ardupilot #23259: Add support for Covariance for GPS Accuracy

Ryanf55, 2023-03-20:
## Feature request

In order to support the ROS2 [NavSatFix](https://docs.ros2.org/galactic/api/sensor_msgs/msg/NavSatFix.html) message, which Ardupilot can publish over DDS, the covariance field for accuracy needs to be filled in. Currently,  `AP_GPS`  does not have any support for covariance; only hdop and vdop. 

**Describe the solution you'd like**

1. Add a generic method to AP_GPS to retrieve the covariance of the current GPS position for each sensor in the front end
2. If a backend GPS cannot supply any accuracy, then there share be an enumeration to say that.
3. If a backend GPS supports hdop, vdop, use it to compute the diagonals of covariance, and an enumeration to say that. 
4. If a backend GPS supports covariance, use it to fully fill in the matrix. 

**Describe alternatives you've considered**

I initially implemented this logic in the DDS layer of Ardupilot, but calculating covariance is something the GPS library could be doing. 

**Platform**
[ x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**

* ROS nmea_navsat_driver covariance support:  https://github.com/ros-drivers/nmea_navsat_driver/issues/17
* Formulas: https://core.ac.uk/reader/236623780

[image]

tridge, 2023-04-04:
I think horizontal_accuracy() and vertical_accuracy() would likely be more appropriate, and less maths

Ryanf55, 2023-04-04:
The notes are how to implement this are discussed here. Ready for implementation. 
https://docs.google.com/document/d/1MNCh-qiqblKFaB3a_ZDKPgmsTwBaopdHQuyWSKnw1Zc/edit#heading=h.kmo72m9ssosu

Ryanf55, 2023-04-04:
Thoughts on the following? 

In **AP_GPS.h** add the following enum

```c++
    // Match https://docs.ros2.org/foxy/api/sensor_msgs/msg/NavSatFix.html
    enum class GPS_CovarianceType {
         UNKNOWN = 0  ///< The GPS does not support any accuracy metrics
        APPROXIMATED = 1  ///< The accuracy is approximated through metrics such as HDOP/VDOP
        DIAGONAL_KNOWN = 2 ///< The diagonal (east, north, up) components of covariance are reported by the GPS
        KNOWN = 3 ///< The full covariance array is reported by the GPS
    };
```

Then, add the following generic method to populate a covariance array. Use either a raw array, or `std::array`, since size is known at compile time. Since covariance in SI units in this order is pretty standard, it can be in AP_GPS rather than AP_DDS.
```c++
    GPS_CovarianceType covariance(const uint8_t instance, double cov[]) const WARN_IF_UNUSED
```
OR
```c++
GPS_CovarianceType covariance(const uint8_t instance, std::array<double, 9>& cov) const WARN_IF_UNUSED
```

WickedShell, 2023-04-04:
Septentrio provides a full covariance matrix for us, we actually derive the horizontal/vertical accuracy numbers from it.

Ryanf55, 2023-04-04:
> Septentrio provides a full covariance matrix for us, we actually derive the horizontal/vertical accuracy numbers from it.

Great to know! I'd love to see some of the nice GPS backends override this method to populate the full covariance matrices when possible, however it's out of scope for the initial PR. Ublox M8 also can do it according to their API docs, and that's a pretty widely used device from what I understand.

WickedShell, 2023-04-04:
@Ryanf55 fair enough. For future reference it's here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS_SBF.cpp#L494 and the struct is here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS_SBF.h#L202

Ryanf55, 2023-04-05:
The usage of this new feature will be covered in this ticket: 
https://github.com/ArduPilot/ardupilot/issues/23277

It's next up. I'm proposing to merge this as-is without a consumer to keep PR scope small.

PR #23389 (2023-04-14): AP_GPS: Add covariance output to AP_GPS matching ROS NavSatFix
This PR adds support for populated the double matrix of covariance, which will be consumed in ROS2 in a follow up PR. 

The covariance units are m^2 (it's a variance).

It's only computed when requested. 

Note, I did clarify this is the position covariance, since it seems the GPS receivers can also report velocity covariance. Currently, the velocity covariance is not part of the NavSatFix(yet), but we can expect a future with a `velocity_covariance` method when that is true.

Closes #23259
closed 2023-04-14 by tridge