ArduPilot/ardupilot #23278: AP_DDS: Add support for PoseStamped

Ryanf55, 2023-03-22:
While #23277 adds support for WGS-84 position reporting, the PoseStamped message is used to report position in a local coordinate system following ENU convention for ROS2 consumers. 

This is commonly used as the position component an Odometry message in ROS2, when combined with the appropriate orientation message.

pedro-fuoco, 2023-04-13:
I'd like to look into it. Can you assign me ? @Ryanf55 
For this first PR should I add support for EKF2, EKF3 and ExternalAHRS ? From what I understand I would need to extract the NED data from one of them and convert it to ENU depending on which one the vehicle is currently using

Ryanf55, 2023-04-24:
Just talked to @tridge. We are good to continue with only ENU support. Since the conversion is usually trivial, and most ROS community is already in ENU, ArduPilot will expose ENU.

If we need NED, ArduPilot should follow ROS REP 103/105/147 by adding the `_ned` suffix to the frame ID. Perhaps enabled with a parameter so there isn't duplicate data.  Not a priority for now unless someone makes a strong case for it.

PR #23480 (2023-04-25): AP_DDS: Local Pose Publisher
Resolves #23278, here is the PR working on SITL:
[image]
The NED to ENU conversion seemed fine when I analysed the position coordinates of a plane flying in SITL. As you can see in the screenshot the plane went North-West after takeoff which resulted in a negative X axis (points towards east in ENU) and a positive Y axis (points towards north in ENU). Z is positive after takeoff.
closed 2023-04-25 by tridge