ArduPilot/ardupilot #23279: AP_DDS: Add support for geometry_msgs/TwistStamped

Ryanf55, 2023-03-22:
Add support for publishing the velocity of the vehicle in the local coordinate system. This message goes hand in hand with #23278 (they are published at the same rate at the same time). 

The Twist message captures the linear and angular velocities of the vehicle following ROS REP 103 conventions (SI units, ENU, right-handed quaternion orientation)

pedro-fuoco, 2023-04-14:
I can have this one. I will work on it as soon as  #23480 merges, as they're quite similar

pedro-fuoco, 2023-04-18:
Should we use a body-fixed or local NED orientation for this topic ? 
MAVROS uses a body-fixed orientation in their [local position plugin](https://github.com/mavlink/mavros/blob/a1dd03c6db26107f914b4526305abca9733f7486/mavros/src/plugins/local_position.cpp#L123), and I agree with them, it is more convenient in velocity data to have an x-forward standard

PR #23541 (2023-04-27): AP_DDS: Local Velocity Publisher
Resolves #23279 
Here is the PR working in its current state in SITL:
[image]
It is currently following:
```
// In ROS REP 103, it follows this convention
// X - Forward
// Y - Left
// Z - Up
// https://www.ros.org/reps/rep-0103.html#axis-orientation
```
For linear velocity, the data is earth-fixed but starts alligned with the robot body (x-forward)
For angular velocity, the gyro data is body-fixed.
closed 2023-04-27 by tridge