ArduPilot/ardupilot #23280: AP_DDS: Support custom build server enabling DDS

Ryanf55, 2023-03-22:
The custom build server has the ability to build custom firmware. Since users may want to add DDS support to their firmware, without compiling on their own computers, it is ideally added as an option to the custom firmware builder. 

See the ArduPilot Dev Call 2023-03-20 notes for more info: 
https://docs.google.com/document/d/1DPtwTY-D-KeboL2WmKkYTOYxiZDSPSYEFVfH-kpkue0/edit#heading=h.kdoemjxcl7ua

pulak-gautam, 2024-01-22:
Hi, I read the source code of custom server. I wanted to confirm that just need to add just the appropriate entry in `BUILD_OPTIONS`, or are there changes need to be made under `ardupilotwaf` as well.
Thanks

Ryanf55, 2024-01-22:
> Hi, I read the source code of custom server. I wanted to confirm that just need to add just the appropriate entry in `BUILD_OPTIONS`, or are there changes need to be made under `ardupilotwaf` as well. Thanks

Yep, please give that a whirl and see what happens.

pulak-gautam, 2024-01-24:
@Ryanf55 I ran the custom build server locally with modified BUILD_OPTIONS, encountering a redefined error for AP_DDS_ENABLED. The back-end commands with verbose showed that waf is configured with -DAP_DDS_ENABLED=0 by default, unlike other libraries in the custom build server, causing a compiler warning.
I attempted to locate where this flag is initialized but couldn't find it. Could you kindly give me some direction?
Thanks

pulak-gautam, 2024-02-27:
It made the necessary changes to prevent conflicts with the definition of AP_DDS_ENABLED in hwdef.h.
The compiler is now unable to recognize any definition of AP_DDS_ENABLED, even though it's clearly defined in hwdef.h generated by the custom build sever. 
It does seem that the hwdef.h (created when we configure waf using extra-hwdef) included by default. Could you let me know if this is correct or not?
Thanks

pulak-gautam, 2024-03-05:
Hi @Ryanf55, after fixing the missing includes and making appropriate changes so that waf would include and recurse the directories. I encountered linker errors for generated Idl msg files. I believe they should have been linked here ``  bld.env.AP_LIB_EXTRA_SOURCES['AP_DDS'] += [str(gen[1])]`` under AP_DDS/wscript. 
I am not sure where I am going wrong. Could you give me some direction? 
Thank you

Ryanf55, 2024-04-16:
> The compiler is unable to recognize any definition of AP_DDS_ENABLED, even though it's clearly defined in hwdef.h generated by the custom build sever. Edit: Fixed by adding the missing includes similar to [#26342](https://github.com/ArduPilot/ardupilot/pull/26342)

Can you contribute the fixes for those?

pulak-gautam, 2024-04-19:
> Can you contribute the fixes for those?
Hi, I am really sorry for the delay. I will be sure to put up a pull request with requested changes for this ticket by next week.
The issues I mentioned earlier were fixed after I raised them on discord (#26405 #26342). I will be sure to make PRs if I face any similar hiccups.

Ryanf55, 2024-10-19:
Related to #26268 where we want to expose all the options.