ArduPilot/ardupilot #23295: Arducopter_Simulink: ''Body Frame Rates -> DCM''  subsystem does no get same result as measure data

MarcLeclercGit, 2023-03-22:
## Bug report
 The subsystem Body Frame Rates -> DCM seems to have a error when 
**Issue details**
Hi,

Using the simulink/adrucopter in the Tools repository I tried to validate that the simulink model would give me the same result as the Autopilot when using the same measured data.  When fed the measured desired roll rate/ pitch rate and yaw rate, the ''Rate Controller'' subsystem give the correct RATE out. When fed measured pitch angle data (simIn.param.attCtrlVal = true), the Attitude controller subsystem also  generate similar desired pitch and yaw rates as the measured one to use as input in the RATE controller. 

However, if I use the ''Body Frame Rates -> DCM '' subsystem and set the switch to use measured angRateBf for the operations from the sequence ''AP_AHRS_DCM::matrix_update() and Matrix3<T>::rotate()'' on it, I do not get the same ''attitudeBodyAct'' and ''eulerAngleAct'' as the measured data. I linked a image to where I did the comparison.
[image]

The ''eulerAngleAct'' for Sim and Meas follow roughly the same trend, but the yaw sign is reverse

[image]

When I also compare the ''attitudeBodyAct'' the quaternions values from the Sim and Measured data follow the same shape but there's a significant offset between the values.

[image]

I linked the flight log that I use for the comparison.I used the subflight when I was in AltHold.

Thanks

**Version**
Master branch

**Logs**
[Log_LineDroneJr_STAB_ALTHOLD_LOIT1.zip](https://github.com/ArduPilot/ardupilot/files/11042147/Log_LineDroneJr_STAB_ALTHOLD_LOIT1.zip)

rmackay9, 2023-03-23:
This is one for @amilcarlucas or @fbredeme I think.

fbredeme, 2023-03-29:
@MarcLeclercGit 
I tried the same setup (feeding in the measured body rates into the "Body Frame Rates -> DCM " subsystem) and I actually get quite satisfying results different from those deviations that your plots show. Can you post a screenshot that shows how you input the measured body rates?

MarcLeclercGit, 2023-03-30:
@fbredeme 
Here's the screenshot on how I was feeding the measured body rates.
[image]

fbredeme, 2023-03-31:
I'm using the same setup as you do and I'm getting completely different signals, as you can see in the following plot for the euler angles (I put the scope at the exact same spot as you do):

[image]

What's your setting for simInMode?

fbredeme, 2023-04-03:
Yes you are right, I used the wrong subflight. But still, my results are different from yours. I can't open the model unfortunately, could you convert it to MATLAB version 2021a?

MarcLeclercGit, 2023-04-03:
Ok, here's the Simulink code in the 2021a version
[arducopter.zip](https://github.com/ArduPilot/ardupilot/files/11138285/arducopter.zip)

fbredeme, 2023-04-05:
Thanks for that. I tested your model for the loiter flight mode and the similarity between measured and simulated signals is still pretty satisfactory I'd say:
Euler angles:
[image]

Quaternions:
[image]

The only thing I changed was using the "old" motor mixer subsystem since a parameter in the new one was missing in my model. I guess that is because you changed something in the sim_init.m.

It appears to me that something with your yaw angle is messed up. Did you change something for the initialization of the "ATT" signals in the sim_init.m? These are used to initialize the DCM matrix that is in return used in the Body Frame Rates -> DCM subsystem to calculate the euler angles and the quaternion.

MarcLeclercGit, 2023-04-06:
I re-downloaded the repo from the master branch to take the original without any changed but I still obtain the same difference for the angles. I only added in the function folder ''fromAxisAngle.m'' and ''quatMult.m'' that were missing when running ''sid_sim_init''.  I don't understand why our results are different.

[image]
[image]

I attached the files ''sid_pre_repo.m'', ''sid_sim_init_repo'' and ''arducopter_repo_R2021a'' that I used. The only thing that I changed from the cloned repository is that I added the rotation order 'xyz' when the function 'dcmFromEuler' is called in ''sid_sim_init_repo.m'' since the function is expecting the 'order' argument. 
[sid_pre_repo.zip](https://github.com/ArduPilot/ardupilot/files/11171194/sid_pre_repo.zip)

fbredeme, 2023-04-21:
I believe I know what causes the problem. In the 4.3 version of the simulink model, I forgot to add the functions "dcmFromEuler.m", ''fromAxisAngle.m'' and "quatMult". This was fixed for the version on the master branch. My "dcmFromEuler.m" does not possess any 'order' argument, so I don't know which function you are using. In my function, which is directly derived from the Ardupilot function "Matrix3<T>::from_euler", the rotation order is zyx. So try using my function instead of the one that you are using now.

MarcLeclercGit, 2023-04-21:
I don't see the functions "dcmFromEuler.m", ''fromAxisAngle.m'' and "quatMult" in the 'functions' folder of the master branch.
Is it because it has not being merged yet?

fbredeme, 2023-04-24:
Yes your are right, I thought that I already created a merge request for that matter. I will catch up doing that asap.

fbredeme, 2023-05-09:
@MarcLeclercGit The merge request has just been approved and closed. The simulink version in the master branch should now work.

amilcarlucas, 2023-12-04:
@MarcLeclercGit Can this be closed now?

MarcLeclercGit, 2023-12-04:
Yes, you can close it.

amilcarlucas, 2023-12-06:
No longer relevant

closed 2023-12-06 by amilcarlucas