ArduPilot/ardupilot #23308: Copter: a bug in AC_AttitudeControl::attitude_controller_run_quat(): limit doesn't satisfy SQRT controller

luweiagi, 2023-03-23:
code: https://github.com/ArduPilot/ardupilot/blob/6839f6c89b51aa94a3961c0ea03470e84409b5f8/libraries/AC_AttitudeControl/AC_AttitudeControl.cpp#L772
'''
    // Limit Yaw Error based on maximum acceleration - Update to include output saturation and maximum error.
    // Currently the limit is based on the maximum acceleration using the linear part of the SQRT controller.
    // This should be updated to be based on an angle limit, saturation, or unlimited based on user defined parameters.
    Quaternion yaw_vec_correction_quat;
    if (!is_zero(_p_angle_yaw.kP()) && fabsf(attitude_error.z) > AC_ATTITUDE_ACCEL_Y_CONTROLLER_MAX_RADSS / _p_angle_yaw.kP()) {
        attitude_error.z = constrain_float(wrap_PI(attitude_error.z), -AC_ATTITUDE_ACCEL_Y_CONTROLLER_MAX_RADSS / _p_angle_yaw.kP(), AC_ATTITUDE_ACCEL_Y_CONTROLLER_MAX_RADSS / _p_angle_yaw.kP());
        yaw_vec_correction_quat.from_axis_angle(Vector3f{0.0f, 0.0f, attitude_error.z});
        attitude_target = attitude_body * thrust_vector_correction * yaw_vec_correction_quat;
    }
'''

rmackay9, 2023-03-23:
Thanks for the report.

We get similar queries about the attitude and position controller fairly regularly and most often they aren't actually bugs but instead are just a misunderstanding of the code.  In any case, I'll leave this one for @lthall to reply to.

luweiagi, 2023-03-24:
> Thanks for the report.
> 
> We get similar queries about the attitude and position controller fairly regularly and most often they aren't actually bugs but instead are just a misunderstanding of the code. In any case, I'll leave this one for @lthall to reply to.

Aha, I see. Thanks for your reply. Maybe I am wrong but at present I think it may be right and fell anxious without any confirmation. Sorry to disturb you.

lthall, 2023-03-24:
You are correct in your assessment of the code.

However, this part of the code needs more work as the comments above suggest. This limit should be based on the position error that would saturate the output.

If we assume saturation with static zero yaw rate and linear controllers for example it would be:
1/(_p_angle_yaw.kP() * get_rate_yaw_pid().kP())

With the square root controller it would be something like:
1/(inv_sqrt_controller(_p_angle_yaw.kP(), bla, bla) * get_rate_yaw_pid().kP())

It will limit the angle to something like 10 to 30 degrees.

@luweiagi Thank you for your effort in highlighting problems that you see in the code.

lthall, 2023-03-24:
#23321

closed 2024-01-24 by rmackay9