ArduPilot/ardupilot #23320: Remove mavlink enum dependencies on non related library

khancyr, 2023-03-24:
## Feature request

**Is your feature request related to a problem? Please describe.**
In numerous place, we were using mavlink enum. Those are well defined and conveniant but they bring a hard link to mavlink that can be harmfull codewise.

Therefore, we got a recent effort to reduce this dependency : see https://github.com/ArduPilot/ardupilot/pull/23229 . In this PR, we decouple the GPS status from the mavlink enum. That lower our inclusion mess.

**Describe the solution you'd like**
We should now update our codebase to remove the mavlink GPS enum in favor of the GPS_Status enum.

Some other work could be done to make the same for EKF and other mavlink enums !

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

khancyr, 2023-03-24:
example of file that need upgrade : https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_ExternalAHRS/AP_ExternalAHRS_LORD.cpp

kar-hik, 2023-03-25:
Can you assign me this issue, I can work on this?

SANTHOSH-MAMIDISETTI, 2023-06-09:
@kar-hik  @khancyr  is this issue still being worked on ?

peterbarker, 2023-09-12:
@kar-hik @SANTHOSH-MAMIDISETTI anything in here is available to be worked on if you feel like it - just read the comments and make sure nobody else has piped up to say they're working on it.

hmoyen, 2024-03-06:
hi! are there any other libraries that need to be upgraded?

Theonewhomadethings, 2024-03-26:
Is there still room to contribute on this issue?, looking for a first issue to be assigned. @peterbarker @khancyr

litski0, 2025-10-22:
update @amilcarlucas , @khancyr  , @peterbarker 
it wasnt only the colcon build which failed 
my pr also failed sitltest for quad plane and copter test
i would like your help with it  PR(#31359 )