ArduPilot/ardupilot #23363: AP_DDS: Design the ROS2 control interface

Ryanf55, 2023-03-30:
For a ROS2 offboard computer looking control the autopilot, there are a variety of methods. 

From a high level, it could be waypoint-based, feeding waypoints and letting the autopilot figure out how to get to them. 

Other control methods (acceleration, rate) control are lower level, and require more real time behavior offboard. Pending some performance metrics that need to be completed, design the generic API for ROS2 to control ardupilot. 

ROS REP-147 exposes a good set of messages, but they don't have any feedback on success of commands, or progress indications. 

Since the DDS interface doesn't support services or actions (yet), this ticket can be scoped on the open loop control.

Ryanf55, 2023-05-24:
Starting to refine this. I'll enclose a proposal below. Here's the rough plan of action:

1. Define the name for control library, apparently Randy is good at naming it. For now: `AP_ExternalControl`
2. Define the coordinate systems and messages for control
  * Mavlink messages will be unchanged, but must be compatable
  * ROS2 can follow REP 147
  * Internal library should not use centidegrees or cm/s, prefer SI
  * It is ok to work in degrees or radians internally, just pick one
4. Draft up the header file for `AP_ExternalControl` within scope for PVA control
    * Waypoints - out of scope

slim71, 2023-06-07:
Some (hopefully useful) feedback.
<br><br>

I've reviewed **REP147** with control algorithms in mind, not only for the project I'll build but also from a general point of view.
I'd say the most important stuff, which should take higher priority, is
1.  the [pose interface](https://ros.org/reps/rep-0147.html#pose-interface), using messages of type `geometry_msgs/PoseStamped` on topic `command_pose` ⇨ this would allow the simplest control (which I guess is maybe the most desirable to the average user): provide a pose to the vehicle and let its autopilot reach it autonomously
2. the [rate interface](https://ros.org/reps/rep-0147.html#rate-interface), with messages `geometry_msgs/TwistStamped` on topic `command_velocity` ⇨ as **REP147** states for ROS, this is one standard way to control robots (and so vehicles too). For example, it could be extremely useful for "manual" controls (i.e. as if a joystick is used)
3. the [odometry info](https://ros.org/reps/rep-0147.html#odometry-interface), using messages `nav_msgs/Odometry` on topic `odom` or the proposed and more complete `nav_msgs/OdometryWithAcceleration` on topic `odometry` ⇨ data estimation is crucial for any real application
<br>

The following would instead add information that is not strictly crucial, but really useful to add more functionalities
- messages `nav_msgs/GlobalPosition` on topic `command_gps_pose` ⇨ I can see many applications using this to specify a mission consisting of many goals. This is also a more granulated way of giving desired poses as goal, with relaxed constraints
- datatypes `mav_msgs/FlightMode` and `mav_msgs/SetFlightMode` for the `set_flight_mode service`

Finally, some more stuff for additional debug purposes or even more functionalities:
- [battery status](https://ros.org/reps/rep-0147.html#battery-status)
- [vehicle status](https://ros.org/reps/rep-0147.html#vehicle-status)
- [IMU](https://ros.org/reps/rep-0147.html#imu-messages) data
- [GPS](https://ros.org/reps/rep-0147.html#gps-messsages) data
- [position reporting](https://ros.org/reps/rep-0147.html#position-reporting) in different frames
- [mission management](https://ros.org/reps/rep-0147.html#mission-management)
- [collision detection](https://ros.org/reps/rep-0147.html#collision-information)
- [connectivity checks](https://ros.org/reps/rep-0147.html#connectivity-checks)

I realize this last list is a bit long, but as I said in my opinion these would have lower priority compared to the ones above.

<br><br>

A couple more things.

> * Internal library should not use centidegrees or cm/s, prefer SI
> * It is ok to work in degrees or radians internally, just pick one

1. I completely agree on using SI units, since it's _universal_. That would also settle the _degree vs radians_ argument, but the important stuff is being consistent. The same applies for _Euler angles vs quaternions_ when it comes to rotations, but I guess that's not an issue here. 
2. Just as an added info, in my case the **_auto_** mode would be crucial, since I plan to control a fleet later on. I think it is also very useful to the general user in both simulations and real applications (just think about "follow me" missions)
3. The update rates would change drastically depending on hardware/simulation, application, vehicle velocity, etc... As a general rule of thumb, average applications should be good with a frequency in the **20Hz-100Hz** range (with 20Hz and a max velocity of 0.5m/s, the vehicle would move 0.025m=2.5cm between two updates and I think that's fairly acceptable)

Ryanf55, 2023-06-08:
# Controls interface

## Position

* In mavlink copter interface, you command a position, and ArduPilot handles the orientation for you. If you don't send velocity or acceleration, it will fly to it, and stop there. The same command handles both position and velocity. 
  * https://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html#copter-commands-in-guided-mode-set-position-target-local-ned
  * https://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html#set-position-target-global-int
* Decision: While mavlink supports local and global position controls, we decided to expose only the local control method in the DDS interface. 

## Velocity Controller
* You can do velocity in ArduPilot, using the same mavlink packets above, see `vx` field for example. The mavlink packet also lets you send a yaw rate, but does not expose roll rates or pitch rates (these are controlled at a different level by ArduPilot itself). 
* If a ROS user is sending velocity controls over `geometry_msgs/TwistStamped`, we should provide documentation on what fields to zero depending on the vehicle type. Randy thinks a level is missing. If we provide a velocity with rotation target but no angle target, there are an infinite number of solutions

## Accerlation Controller

* We can also implement this, but it's likely going to be used less than position and velocity, so less of a priority.

## Trajectory and Waypoint Control

Let's skip these for now (at least in scope of GSoC)

Ryanf55, 2024-11-21:
Pretty much all the main control interfaces are implemented that our users have requested. The interface seems pretty stable, and we haven't found any design issues, so I'd say this interface is now complete. 

The use of the AP_ExternalControl library is crucial to keep this interface simple, clean, and documented.