ArduPilot/ardupilot #23386: Copter: attitude_error from axis_angle in thrust_vector_rotation_angles() may have problem

luweiagi, 2023-04-04:
Copter: function AC_AttitudeControl::thrust_vector_rotation_angles( ) may be wrong.

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_AttitudeControl/AC_AttitudeControl.cpp#L780

```
    // calculate the angle error in x and y.
    Vector3f rotation;
    thrust_vector_correction.to_axis_angle(rotation);
    attitude_error.x = rotation.x;
    attitude_error.y = rotation.y;

    // calculate the remaining rotation required after thrust vector is rotated transformed to the body frame
    // heading_vector_correction
    Quaternion heading_vec_correction_quat = thrust_vector_correction.inverse() * attitude_body.inverse() * attitude_target;

    // calculate the angle error in z (x and y should be zero here).
    heading_vec_correction_quat.to_axis_angle(rotation);
    attitude_error.z = rotation.z;
```

lthall, 2023-04-04:
Yes, this is an interesting decision I made. This is done as a two rotation system. The first removes error in roll and pitch, then the second removes the yaw error afterwards. This is for two reasons, roll and pitch is both the most important control axis and the fastest to respond. Second is we must expect to loose control of yaw completly during a motor loss. This approach lets me prioritise roll and pitch attitude control over yaw, without compromise.

So to try to summarise. We control roll and pitch like there is always zero yaw error, or we don't care about yaw angle. Then we control yaw like there is no roll and pitch error under the assumption that yaw is much slower or less accurate than roll and pitch.

luweiagi, 2023-04-04:
> Yes, this is an interesting decision I made. This is done as a two rotation system. The first removes error in roll and pitch, then the second removes the yaw error afterwards. This is for two [...]

Thank you for the very patient and affable reply! 

Now I figure out why it is always said that there is a sequence: first roll/pitch and then yaw. I used to think it was an instantaneous mathematical process.

Now I kown the reason is roll/pitch change is fast because motor force derectly change roll/pitch, and yaw change is slow because it relys on the torque difference of motors which is indirectly and not enough. So it creats a physical-existing sequence of "roll/pitch change first and yaw change second".

lthall, 2023-04-04:
>> @lthall Thank you for the very patient and affable reply!

Mate, I am very appreciative of your willingness to read through the code so thoroughly and bring anything that doesn't look quiet right to my attention. People like yourself make a very significant contribution to this code base!!

You are more than welcome to reach out to me for any help you need, or questions you have!

esaldiran, 2023-04-05:
@luweiagi There are multiple academical study that goes deep into mathematical derivation of this approach such as [Tilt-prioritized quadrocopter attitude control](https://ieeexplore.ieee.org/abstract/document/8556372). I hope you find it helpfull.

lthall, 2023-04-05:
@esaldiran Thanks for that link. The abstract of that paper does seem to describe something similar to the approach I took here. However, none of my work has been inspired by, or based on, any academic paper so I am not sure how close the math of this approach and my approach are.

luweiagi, 2023-04-10:
> 

Thanks! I have download this paper http://brescianini.info/wp-content/uploads/publications/breTCST18.pdf, and will read it when I have time.
> @luweiagi There are multiple academical study that goes deep into mathematical derivation of this approach such as [Tilt-prioritized quadrocopter attitude control](https://ieeexplore.ieee.org/abstract/document/8556372). I hope you find it helpfull.

luweiagi, 2023-04-10:
> Mate, I am very appreciative of your willingness to read through the code so thoroughly and bring anything that doesn't 

@lthall  Thanks, I feel interesting and happy to try to understand how anthor thinks when he is writing code.

There is a question I have never understood for several years. I haven't found a suitable opportunity to consult you. This problem has been bothering me for a long time. I am very curious about it and want to know why you write this code.

In function attitude_controller_run_quat():
https://github.com/ArduPilot/ardupilot/commit/ad6e2eeed838e734fba5c946874e2bd2ef3ad1ed

[image]

lthall, 2023-04-10:
This was something I was unsure if it was good or bad and I eventually decided it was the wrong thing to do.

In the case of lost yaw control this keeps the roll and pitch angle error constant-ish so that the attitude controller doesn't need to be as fast. My concern was that if the aircraft was rotating at say 1 Hz or faster the latency in the attitude controller could result in the corrections being out of phase.

I tried a number of times over the years to establish if it made things better or not but I couldn't satisfy myself that it was the right way to go. I couldn't justify it, it is not clear to others what it is doing, I removed it.

lthall, 2023-07-23:
I think we are fine to close this now @luweiagi. Let me know if I should reopen.

closed 2023-07-23 by lthall

lixiaowei123, 2024-02-21:
> @esaldiran Thanks for that link. The abstract of that paper does seem to describe something similar to the approach I took here. However, none of my work has been inspired by, or based on, any academic paper so I am not sure how close the math of this approach and my approach are.
Hello, I would like to know how you came up with this method at that time. Is there no reference available