ArduPilot/ardupilot #23387: AP_DDS: Support external odometry sources

pedro-fuoco, 2023-04-04:
## Feature request

Add support in DDS for subscribing to external odometry sources 

**Describe the solution you'd like**
AP_DDS Client should receive the Odometry data from the MicroROS Agent and feed Ardupilot's Extended Kalman Filter with it. Here's a diagram of the proposed architecture using cartographer_ros as an exemple of external odometry source:
[image]

**Describe alternatives you've considered**
This is a DDS alternative to /mavros/vision_pose/pose

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

Ryanf55, 2023-04-04:
Let's remove the construct that external pose can come from vision when this is implemented. External pose could also come from LiDAR, or wheel encoders, or even an external GNSS + INS.

pedro-fuoco, 2023-05-05:
Initial tests can be made with rosbag2 
https://google-cartographer-ros.readthedocs.io/en/latest/demos.html

pedro-fuoco, 2023-05-08:
By default cartographer_ros publishes a tf tree instead of a position topic. 
For the [ROS integration in GSoC 2022](https://github.com/ArduPilot/ardupilot_ros/blob/main/launch/cartographer.launch) an additional ROS package was used, "robot_pose_publisher", to get the base_link->odom transformation and publish it into a position topic, which was then retrieved by mavros and inputted at Ardupilot's EKF

I'm investigating a way to bypass this

pedro-fuoco, 2023-05-16:
Jitter correction
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_VisualOdom/AP_VisualOdom.h

Ryanf55, 2023-05-16:
As part of this work, we would need to work with jitter time correction from transport + processing lag. 

> Jitter correction https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_VisualOdom/AP_VisualOdom.h

Ryanf55, 2023-08-06:
To add more detail: 
Localization data from cartographer will be from `odom` to `base_link`. 

I read the research paper [Localization strategies for autonomous mobile robots: A review](https://www.sciencedirect.com/science/article/pii/S1319157821000550)

They detailed three types of localization:
* Position tracking
* Global positioning/localization
* Kidnapped robot problem

This PR will support the first type - where the initial pose in global coordinates is known, then the position of the robot is augmented with external data, which will come from the companion computer. Although this position estimate has uncertainty, the current ROS interfaces do not expose any accuracy fields.  Can @rmackay9 help me understand how the accuracy data was retrieved from the SLAM system and shared to ArduPilot when using Mavlink?

Ryanf55, 2023-12-06:
Completed by  #24518.
Automated testing covered in #25711