ArduPilot/ardupilot #23392: COMPASS_ORIENT 38

bajbaa, 2023-04-04:
## Bug report

**Issue details**
COMPASS_ORIENT = 38, wrong Roll angle

_Please describe the problem_
This 38 is disiagned for 3DR SOLO stock external compass and has wrong Roll angle. Should be 90° instead of 180°
is Yaw293Pitch68Roll180 but should be Yaw293Pitch68Roll90

**Version**
All versions

**Platform**
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
3DR SOLO

**Hardware type**
Black Cube, stock external compass HMC5883 in leg

**Logs**
drone don't want to arm because compasses inconsistent. I had checked it on three difrent 3DR SOLO drones, also measured angles from 3D model of 3DR. I solved the problem using:
COMPASS_ORIENT = 101
CUST_ROT_ENABLE = 1
CUST_ROT1_PITCH = 68
CUST_ROT1_ROLL = 90
CUST_ROT1_YAW = 293
but would be better for all users of 3DR Solo drone to fix COMPASS_ORIENT = 38 ;)

davidbuzz, 2023-04-04:
Please provide more info about your specific setup, and the testing that was done

bajbaa, 2023-04-06:
I have got four pieces of 3DR Solo drone. All have a stock external compass in the leg. After upgrading my drones to OpenSolo and AutoPilot I had a problem with compasses. I always got the message "compasses inconsistent" and had to do compass calibration and could fly only once. When I changed the battery it happens again. Because I have four drones I knew that it could not be a broken external compass. Of Course  I did some testing like disconnecting external compass and other tests and I saw that external compass showed the wrong azimut in Mission Planner. Also differences between both compasses at mx my and mz were too big.

Then I downloaded a 3D model of the 3DR Solo drone and measured Yaw, Pitch and Roll angles in CAD software. My results you can see on screen shots. So this matches almost with COMPASS_ORIENT = 38 instead of Roll angle. This mistake is even visible by eyes if someone has a good 3D imagination ;)

Photos and screenshots:
[https://drive.google.com/drive/folders/1jZfWQUiG_7XA9Z-DioDcJB3PIM1oFCj8?usp=sharing]

I went around this error using custom COMPASS_ORIENT = 101 and left information for users of 3DR Solo drone on 3drpilots.com forum how to do it.
[https://3drpilots.com/threads/compasses-inconsistent.16241]

Please fix this bug to let others use 38 instead of 101 COMPASS_ORIENT.

bajbaa, 2023-04-06:
One of my tests was to unscrew the outer compass and attach it with duct tape underneath the drone body with all angles equal to zero and it works as it should! No more problems with "compasses inconsistent". So I knew that something was wrong iwith compass orientation ;) Then I checked the angles on 3D model :)

markingle, 2023-04-10:
@davidbuzz I am struggling with this issue too. My project involves modifying 3DR Solo into a drone for drone shows and I'm making good progress. I am using a custom version of ArduPilot that I have modified make memory available and enable LED control through AP_Notify. If you can recommend some direction I can attempt to resolve this issue with help from @bajbaa

JBB2068, 2025-05-19:
@rmackay9 @davidbuzz 
It seems like the routine for preserving the user settings may be causing this. Is that a possibility?

It happens mostly when updating from the old 3dr branch of Copter.

I believe the actual problem here is not the orientation. During an update to the Cube firmware, the compass order is being scrambled. It seems inconsistent but it is happening quite a lot lately. The orientation 38 is being moved to COMPASS3 while COMPASS and COMPASS2 are set to 0. This sometimes causes the compass calibration to fail repeatedly, but other times if COMPASS_AUTO_ROT is set to 2 it will change COMPASS_ORIENT to 29 and the calibration will complete. But after a reboot it shows Compasses Inconsistent.

If I change the parameters to
COMPASS_ORIENT 38
COMPASS2_ORIENT 0
COMPASS3_ORIENT 0
and reboot, Mission Planner tells me the compasses have changed, and shows them at the top of the Compass page in the wrong order. So I arrange them correctly and redo the compass calibration and reboot again and this solves the problem.