ArduPilot/ardupilot #23410: Copter skips first waypoint if in auto without position and no takeoff command

peterbarker, 2023-04-06:
## Bug report

**Issue details**

If you mission starts with a NAV_WAYPOINT command and you enter auto before the vehicle has set its origin then the vehicle will immediately skip to the second waypoint as soon as the origin is set.

If the first waypoint is a takeoff, this problem doesn't occur.

Randy theorises that we don't run the scurve navigation code when the first item is a takeoff, but do if it the first item is a waypoint command.  The spline calculations are not coping with the fact we don't have a position and are somehow assuming we are at our destination.

Possible fix is to not run the mission state machine until we have a position for the first time.

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
+

**Hardware type**
SITL

**Logs**

snktshrma, 2023-04-09:
Hi @peterbarker! I am trying to recreate this issue on SITL on my system but I think I am mission out something. Can you repeat how to recreate it?

Thanks

notken12, 2024-10-17:
I'm experiencing the same issue but it skips the first waypoint regardless of whether it's a NAV_WAYPOINT or takeoff command.