ArduPilot/ardupilot #23416: Parameter changes brick flight controller

evaesteban, 2023-04-06:
## Bug report

**Issue details**

I’m using the Holybro Kakute H7 V2 with ArduPilot and connecting the VN-100 AHRS system to it. When I set the parameter EAHRS_TYPE to 1 (VectorNAV) using QGroundStation the flight controller stops responding. It gets disconnected and I have to erase the bootloader and flash it all again to get it back. This is also happening to me when changing some of the initial tuning paramters for the first flight. I tried it on two different Kakute H7 V2 boards with the same outcome.

**Version**
4.3.5

**Platform**
Copter

**Airframe type**
Octa quad

**Hardware type**
Kakute H7 V2

**Logs**

andyp1per, 2023-04-06:
There was a bug in the bootloader layout of flash parameters in earlier versions - are you able to verify that this problem still happens if you use the latest 4.3.5 bootloader?

evaesteban, 2023-04-06:
Could you send me the link to test and check that I am using the correct one?

peterbarker, 2024-07-31:
@evaesteban OK, so that "time-sensitive" thing really did not work out.  We're working to clear out our issues list so this sort of thing doesn't happen in the future.

We now have VectorNav as a partner, and  the VN-100 is tested and working.

Apologies for the delayed (useless...) response.

closed 2024-07-31 by peterbarker