ArduPilot/ardupilot #23424: AP_DDS: Investigate running microROS client library

Ryanf55, 2023-04-07:
## Feature request

**Is your feature request related to a problem? Please describe.**
Ardupilot parameters would like to be exposed through the ROS 2 parameter server, but it's not clear how to do it. 

**Describe the solution you'd like**
Conduct a prototype on SITL with the microROS agent to determine if ArduPilot can use it. If so, see if we can add chibiOS support for it, as well as specific hardware. It may be best to start on the ESP32 hardware since that's an already supported configuration.

**Describe alternatives you've considered**

@arshPratap is considering using an integration service through eProsima and exposing params as a ROS service 

**Platform**
[x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**

https://design.ros2.org/articles/ros_parameters.html

This ticket is research for #23309

srmainwaring, 2023-07-20:
## micro-ROS PoC

Initial investigations into using micro-ROS directly in ArduPilot:

- https://github.com/srmainwaring/ardupilot/tree/wips/wip-micro-ros

The branch provides an example of ArduPilot hosting a micro-ROS client.

#### Features

- Added library `AP_UROS`.
- Add build flag `--enable-uros`and defines `AP_UROS_ENABLED`, `AP_UROS_UDP_ENABLED`.
- Add params `UROS_ENABLE`, `UROS_PORT`.
- Add  UROS client `AP_Vehicle`.
- Integrate into build system and `sim_vehicle.py`.
- Add simple subscriber to a `geometry_msgs/msg/Vector3`.

#### Notes:

- Only supported for SITL on macOS / linux.
- Requires an externally built library `libmicroros.a` and include folder to be places in  `libraries/AP_UROS/uros/`.
- A custom UDP transport is provided (adapted from AP_DDS), but is disabled because there are link errors when the static micro-ROS client is built for `RMW_UXRCE_TRANSPORT=custom`.
- With the custom transport disabled micro-ROS the default `udp` is used. This requires the static micro-ROS client be built for `RMW_UXRCE_TRANSPORT=udp`.
- DDS and UROS may not be used at the same time as there are conflicts from the message includes (`std_msgs` etc.).

### Dependencies

A customised version of the [`micro_ros_setup`](https://github.com/micro-ROS/micro_ros_setup) project is used to  [create a custom static micro-ROS library](https://micro.ros.org/docs/tutorials/advanced/create_custom_static_library/)

- https://github.com/srmainwaring/micro_ros_setup/tree/wips/wip-humble-macos-genlib

The main changes are:

- Some tweaks to build on macOS which does not support `rosdep` or `dpkg`.
- Change the build to include debug symbols (still optimised).
- Addition of `tf2_msgs` from the [`geometry2`](https://github.com/ros/geometry2) package.
- Modify `client-colcon-meta` and `demo_toolchain.cmake`.

After following the initial steps to configure the workspace and  build `micro_ros_setup`, the `create_firmware_ws` and `build_firmware` steps are:

```bash
ros2 run micro_ros_setup create_firmware_ws.sh
```

```bash
ros2 run micro_ros_setup build_firmware.sh -f -- $(pwd)/src/micro_ros_setup/config/generate_lib/generic/demo_toolchain.cmake $(pwd)/src/micro_ros_setup/config/generate_lib/generic/client-colcon.meta
```

The contents of the folder `~ws/firmware/build` are copied to `~/ardupilot/libraries/AP_UROS/uros`.

### Build

Build directly using `waf`:

```bash
./waf configure --boards sitl --enable-uros
./waf build --enable-uros
```

Build using `colcon`:

```bash
colcon build --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo --packages-select ardupilot_sitl
```

### Run

If you have a `ros-ardupilot` environment set up the various components may be launched separately:

micro-ROS agent:
[... 106 lines cut ...]
bin/arduplane         2457600    294912        0               2752512  Not Applicable  Not Applicable         
bin/ardusub           2211840    278528        0               2490368  Not Applicable  Not Applicable         
bin/blimp             1900544    278528        0               2179072  Not Applicable  Not Applicable         
bin/ardurover         2293760    294912        0               2588672  Not Applicable  Not Applicable         
```

The UROS build has `rmw_microxrcedds` configured as follows:

```bash
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DCMAKE_BUILD_TYPE=RelWithDebInfo",
                "-DRMW_UXRCE_TRANSPORT=custom",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=127.0.0.1",
                "-DRMW_UXRCE_DEFAULT_UDP_PORT=2019",
                "-DRMW_UXRCE_MAX_NODES=15",
                "-DRMW_UXRCE_MAX_PUBLISHERS=15",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=15",
                "-DRMW_UXRCE_MAX_SERVICES=15",
                "-DRMW_UXRCE_MAX_CLIENTS=15",
                "-DRMW_UXRCE_STREAM_HISTORY=32",
                "-DRMW_UXRCE_MAX_HISTORY=10"
            ]
        }
```

patrickelectric, 2025-04-24:
Zenoh [released 1.0 last year](https://zenoh.io/blog/2024-10-21-zenoh-firesong/), that guarantees a stability and maturity of the API, is also becoming a [tier 1 RMW of ROS2](https://github.com/ros2/rmw_zenoh) and zenoh-pico just merged the [LWIP support](https://github.com/eclipse-zenoh/zenoh-pico/pull/932).

Ryanf55, 2025-04-24:
> Zenoh [released 1.0 last year](https://zenoh.io/blog/2024-10-21-zenoh-firesong/), that guarantees a stability and maturity of the API, is also becoming a [tier 1 RMW of [...]

Nice, do you mind creating an issue for Zenoh support and we can discuss there? I'm curious