ArduPilot/ardupilot #23483: PANIC: Failed to start JSBSim

ParasInternKhushPatil, 2023-04-14:
## Bug report

PANIC: Failed to start JSBSim

Cloned, built and installed jsbsim.
When running SITL with jsbsim using the following command:
cd ardupilot/ArduPlane; sim_vehicle.py -f jsbsim, the xterm shows JSBSim to be running while mavproxy reports Link down

**Version**
Latest Version 

**Platform**
[  ] All
[  ] AntennaTracker
[ * ] Copter
[ * ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Any Model of JSBsim

**Hardware type**
SITL 

**Logs**
PANIC: Failed to start JSBSim
Running: sh Tools/scripts/dumpstack.sh 15035 >dumpstack.sh_arduplane.15035.out 2>&1
dumpstack.sh has been run.  Output was:
-------------- begin dumpstack.sh output ----------------
Could not attach to process.  If your uid matches the uid of the target
process, check the setting of /proc/sys/kernel/yama/ptrace_scope, or try
again as the root user.  For more details, see /etc/sysctl.d/10-ptrace.conf
ptrace: Operation not permitted.
/tmp/gdb.15041:2: Error in sourced command file:
No stack.
-------------- end dumpstack.sh output ----------------
Running: sh Tools/scripts/dumpcore.sh 15035 >dumpcore.sh_arduplane.15035.out 2>&1
dumpcore.sh has been run.  Output was:
-------------- begin dumpcore.sh output ----------------
Could not attach to process.  If your uid matches the uid of the target
process, check the setting of /proc/sys/kernel/yama/ptrace_scope, or try
again as the root user.  For more details, see /etc/sysctl.d/10-ptrace.conf
ptrace: Operation not permitted.
/tmp/gdb.15049:2: Error in sourced command file:
You can't do that without a process to debug.
-------------- end dumpcore.sh output -

khancyr, 2023-04-14:
Hello,

Thanks for this. We are aware of this and will update the instructions soon.

ParasInternKhushPatil, 2023-04-14:
Can you state particular lines of code that cause this issue

ParasInternKhushPatil, 2023-04-26:
Any update @khancyr

peterbarker, 2024-07-31:
Did this problem ever get resolved, @ParasInternKhushPatil ?

peterbarker, 2024-08-02:
OK, sorry we didn't chase this up earlier @ParasInternKhushPatil - hope you got it sorted.  If not, please re-open this.

closed 2024-08-02 by peterbarker

Saleh-At, 2024-11-29:
I have same problem
How to solve this