ArduPilot/ardupilot #23488: AP_GPS_UBLOX init code fails when sending GPS initilisation blob to enter ubx binary mode

fronders, 2023-04-14:
## Bug report

**Issue details**

MatekL431-GPS hwdef uses DMA for USART3 communications with u-blox M10. The firmware when built with `--debug --enable-assert` would freeze right after startup. Attaching a debug probe showed code being stuck in a `chSysHalt()` caused by `chDbgAssert()` in the UARTDriver.cpp `UARTDriver::write_pending_bytes_DMA(uint32_t n)` on line 985. 
There `timeout_us` is calculated to be 100500 given `tx_len` equal to 46 and `_baudrate` of 4800 which overflows for 16-bit timer used as SysTick

https://github.com/ArduPilot/ardupilot/blob/e6d6848fb19e32692228be9cee10c97c9f992f24/libraries/AP_HAL_ChibiOS/UARTDriver.cpp#L982-L988

This results in `chTimeUS2I(timeout_us)` assert fail "conversion overflow" @ line 377

https://github.com/ArduPilot/ChibiOS/blob/1ec9f168b2e64a043470a87225b405e79d187354/os/rt/include/chtime.h#L369-L381

Further investigation showed this was cause by 46-byte payload of AP_GPS `_initialisation_blob[]` (or specifically `UBLOX_SET_BINARY_230400`) being sent to switch u-blox into binary mode.

**Version**
AP_Periph v1.5.0

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine
[x] AP_Periph

**Hardware type**
[MATEKSYS M10-L4-3100](http://www.mateksys.com/?portfolio=m10-l4-3100) with AP_Periph MatekL431-GPS build

**Logs**

`write_pending_bytes_DMA()` frame variables:
[image]

Call stack:
```
bt 10
#0  chSysHalt (reason=reason@entry=0x803183c "chTimeUS2I") at ../../modules/ChibiOS/os/rt/src/chsys.c:190
#1  0x0802359e in chTimeUS2I (usec=100500) at ../../modules/ChibiOS/os/rt/include/chtime.h:377
#2  ChibiOS::UARTDriver::write_pending_bytes_DMA (this=this@entry=0x20004928 <uartDDriver>, n=<optimized out>) at ../../libraries/AP_HAL_ChibiOS/UARTDriver.cpp:985
#3  0x08023728 in ChibiOS::UARTDriver::write_pending_bytes (this=this@entry=0x20004928 <uartDDriver>) at ../../libraries/AP_HAL_ChibiOS/UARTDriver.cpp:1095
#4  0x08023abe in ChibiOS::UARTDriver::_tx_timer_tick (this=this@entry=0x20004928 <uartDDriver>) at ../../libraries/AP_HAL_ChibiOS/UARTDriver.cpp:1334
#5  0x08023b84 in ChibiOS::UARTDriver::uart_thread (this=0x20004928 <uartDDriver>) at ../../libraries/AP_HAL_ChibiOS/UARTDriver.cpp:137
#6  0x08023bd2 in ChibiOS::UARTDriver::uart_thread_trampoline (p=<optimized out>) at ../../libraries/AP_HAL_ChibiOS/UARTDriver.cpp:207
#7  0x0800b376 in _port_thread_start ()
```

fronders, 2023-04-14:
One of the ways I can see to fix this is to limit the chunk of AP_GPS initialisation blob being sent at a time depending on selected baudrate:

```cpp
/*
  send some more initialisation string bytes if there is room in the
  UART transmit buffer
 */
void AP_GPS::send_blob_update(uint8_t instance)
{
    // exit immediately if no uart for this instance
    if (_port[instance] == nullptr) {
        return;
    }

    if (initblob_state[instance].remaining == 0) {
        return;
    }
    // limit size of chunk depending on baudrate to make sure the transmit timeout doesn't overflow
    const uint32_t max_chunk = ((((time_conv_t)TIME_MAX_INTERVAL - 500ULL) * _port[instance]->get_baud_rate()) / 10000000U) - 2;
    // see if we can write some more of the initialisation blob
    const uint16_t n = MIN(max_chunk, 
                           MIN(_port[instance]->txspace(),
                               initblob_state[instance].remaining));
    const size_t written = _port[instance]->write((const uint8_t*)initblob_state[instance].blob, n);
    initblob_state[instance].blob += written;
    initblob_state[instance].remaining -= written;
}
```

The `max_chunk` here is just a reverse calculation from `timeout_us` in UARTDriver.cpp
```cpp
uint32_t timeout_us = ((1000000UL * (tx_len+2) * 10) / _baudrate) + 500;
```

peterbarker, 2025-03-27:
Many thanks for this report.  Really wish we had seen this sooner - tridge spent a whole chunk of time tracking this down - and you'd already done the work.

We merged the fix here: https://github.com/ArduPilot/ardupilot/pull/28241

Again thanks, sorry we didn't get to this sooner!

closed 2025-03-27 by peterbarker