ArduPilot/ardupilot #23509: Baro-less flight controllers achieve very high groundspeed while sitting on the bench

andyp1per, 2023-04-15:
I have two boards (Foxeer reaper and Mamba F405 v2), both without baros and both show steadily increasing groundspeed despite reasonable accel calibration and offsets. The fact that two boards do the same leads me to conclude that this is a software/EKF problem

## Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

alexlysenko92, 2024-12-30:
Hi! Any updates on this?

andyp1per, 2024-12-30:
This was fixed by https://github.com/ArduPilot/ardupilot/pull/24656

closed 2024-12-30 by andyp1per