ArduPilot/ardupilot #23529: GPS_GNSS_MODE parameter is not written to memory of ublox f9p

DavideLentini, 2023-04-17:
## Bug report

**Issue details**

Hi all , this is a problem we found when we tried to use the GPS parameters to overwrite the parameters saved in memory of the ublox f9p.
What we noticed is that among the GPS parameters we modify , only 1 is not passed and overwritten in memory , that is the `GPS_GNSS_MODE `parameter . The operations I did were to change the value of this parameter in ardupilot and write it and then restart the drone. The value you see in the image is only demonstrative , but in reality the parameter is not overwritten even with other combinations 

[image]

Going to check with the u-center software , in the screen related to the configurations of the f9p , the parameters do not reflect the order with which I wrote the parameter `GPS_GNSS_MODE `: 

[image]

I repeat that however this problem seems to be there only for this parameter in fact the other parameters such as :
`GPS_MIN_ELEV `and `GPS_RATE_MS `. Are being changed correctly

I know that in the documentation it was stated that the value of constellations could not be changed by ardupilot :
https://ardupilot.org/plane/docs/common-gps-ublox-firmware-update.html#configure-satellite-constellations

But thanks to the help of @khancyr , he confirmed me that this feature was theoretically added .

Thank you very much for the help to everyone

**Version**
4.3.3
**Platform**
[  ] All
[  ] AntennaTracker
[X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

fronders, 2023-04-17:
I just came to report a similar problem, but with u-blox M10 - currently ardupilot doesn't even apply any of the GPS params. 

The reason is for M10 the firmware doesn't support any of UBX-CFG-*** commands, since they got deprecated and we need to use VALSET \ VALGET instead. The interface description for M10 doesn't even mention any of used UBX-CFG commands like UBX-CFG-GNSS or UBX-CFG-RATE. For M10 instead a hardcoded array of params is applied that always shuts down GLONASS and Airborne 4G model for example. 

This is probably the same for F9P/M9 with latest firmware as their datasheets mention the UBX-NAV-*** is deprecated for firmwares v23.01+

Right now I have a dirty fix working in place of that fixed M10 config, but this definitely needs to be revised on a driver-level

Here's how I quick-fixed it in `void
AP_GPS_UBLOX::_request_next_config(void)`:
```cpp
    case STEP_M10: {
        if (_hardware_generation == UBLOX_M10) {
            // special handling of M10 config
            std::vector<AP_GPS_UBLOX::config_list> config_M10;
            // update config with param settings
            if (gps._gnss_mode[state.instance] != 0) {
                Debug("GNSS Settings:\n");

                bool gps_enable = (gps._gnss_mode[state.instance] & (1 << GNSS_GPS)) != 0;
                Debug("GPS L1C/A %u\n", (unsigned)gps_enable);
                config_M10.push_back({ ConfigKey::CFG_SIGNAL_GPS_ENA, gps_enable});
                config_M10.push_back({ ConfigKey::CFG_SIGNAL_GPS_L1CA_ENA, gps_enable});
                // M10 doesn't support GPS L2C and L5

                bool galileo_enable = (gps._gnss_mode[state.instance] & (1 << GNSS_GALILEO)) != 0;
                Debug("Galileo E1 %u\n", (unsigned)galileo_enable);
                config_M10.push_back({ ConfigKey::CFG_SIGNAL_GAL_ENA, galileo_enable});
                config_M10.push_back({ ConfigKey::CFG_SIGNAL_GAL_E1_ENA, galileo_enable});
                // M10 doesn't support Galileo E5A and E5B

                bool beidou_enable = (gps._gnss_mode[state.instance] & (1 << GNSS_BEIDOU)) != 0;
                Debug("BeiDou B1I %u B1C %u\n", 0U, (unsigned)beidou_enable);
                config_M10.push_back({ ConfigKey::CFG_SIGNAL_BDS_ENA, beidou_enable});
                config_M10.push_back({ ConfigKey::CFG_SIGNAL_BDS_B1_ENA, 0});
                config_M10.push_back({ ConfigKey::CFG_SIGNAL_BDS_B1C_ENA, beidou_enable});
                // M10 doesn't support BeiDou B2 and B2A
                // M10 can't have BeiDou B1I enabled simultaneously with BeiDou B1C or GLONASS L1OF
                
                bool qzss_enable = (gps._gnss_mode[state.instance] & (1 << GNSS_QZSS)) != 0;
                Debug("QZSS L1C/A %u L1S %u\n", (unsigned)qzss_enable, (unsigned)qzss_enable);
                config_M10.push_back({ ConfigKey::CFG_SIGNAL_QZSS_ENA, qzss_enable});
                config_M10.push_back({ ConfigKey::CFG_SIGNAL_QZSS_L1CA_ENA, qzss_enable});
                config_M10.push_back({ ConfigKey::CFG_SIGNAL_QZSS_L1S_ENA, qzss_enable});
                // M10 doesn't support QZSS L2C and L5

                bool glonass_enable = (gps._gnss_mode[state.instance] & (1 << GNSS_GLONASS)) != 0;
                Debug("GLONASS L1 %u\n", (unsigned)galileo_enable);
                config_M10.push_back({ ConfigKey::CFG_SIGNAL_GLO_ENA, glonass_enable});
                config_M10.push_back({ ConfigKey::CFG_SIGNAL_GLO_L1_ENA, glonass_enable});
                // M10 doesn't support Glonass L2
            }
            if (gps._sbas_mode != AP_GPS::SBAS_Mode::DoNotChange) {
                bool sbas_enable = gps._sbas_mode;
                Debug("SBAS L1C/A %u\n", (unsigned)sbas_enable);
                config_M10.push_back({ ConfigKey::CFG_SIGNAL_SBAS_ENA, sbas_enable});
                config_M10.push_back({ ConfigKey::CFG_SIGNAL_SBAS_L1CA_ENA, sbas_enable});
            }
            if (gps._navfilter != AP_GPS::GPS_ENGINE_NONE) {
                Debug("Changing engine setting to %u\n", (unsigned)gps._navfilter);
                config_M10.push_back({ ConfigKey::CFG_NAVSPG_DYNMODEL, (uint32_t)gps._navfilter});
            }
            if (gps._min_elevation != -100) {
                Debug("Changing min elevation to %d\n", (int)gps._min_elevation);
                config_M10.push_back({ ConfigKey::CFG_NAVSPG_DYNMODEL, (uint32_t)gps._min_elevation});
            }
            // require a minimum measurement rate of 5Hz
            uint16_t meas_rate = gps.get_rate_ms(state.instance);
            Debug("Changing gps update rate to %u\n", (unsigned)meas_rate);
            config_M10.push_back({ ConfigKey::CFG_RATE_MEAS, (uint32_t)meas_rate});
            config_M10.push_back({ ConfigKey::CFG_RATE_NAV, 1});
            config_M10.push_back({ ConfigKey::CFG_RATE_TIMEREF, 0});    // UTC time

            const config_list *list = config_M10.data();
            const uint8_t list_length = config_M10.size();
            Debug("Sending M10 settings");
            if (!_configure_config_set(list, list_length, CONFIG_M10, UBX_VALSET_LAYER_RAM | UBX_VALSET_LAYER_BBR)) {
                _next_message--;
            }
        }
        break;
    }
```

DavideLentini, 2023-04-18:
@fronders 
Thanks for the help ! 
Yes I think the best thing is to get the main developers involved in this issue . Also because I discovered it by accident , from all this time I thought that GPS_GNSS_MODE parameter change in ublox f9p , but it is not so apparently ... I can imagine how many people mistakenly think that .

seksitha, 2023-05-09:
I have problem with Qoi-tek CAN-GPS M10.

I need to apply quick fix by: 

````
case STEP_M10: {
        // special handling of M10 config
    if (_hardware_generation == UBLOX_M10) {
        const config_list *list = config_M10;
        const uint8_t list_length = ARRAY_SIZE(config_M10);
        Debug("Sending M10 settings");
        const uint8_t buf[]={0xB5, 0x62, 0x06, 0x8A, 0x23, 0x00,0x01, 0x01, 0x00, 0x00, 0x01, 0x00, 0x21, 0x30, 
        0xC8, 0x00, 0x20, 0x00, 0x31, 0x10, 0x00, 0x22, 0x00, 0x31, 0x10, 0x01, 0x0F, 0x00, 0x31, 0x10, 0x00,
        0x0D, 0x00, 0x31, 0x10, 0x01, 0x21, 0x00, 0x11, 0x20, 0x08, 0x8D, 0xCD};
        port->write(buf,sizeof(buf));
        if (!_configure_config_set(list, list_length, CONFIG_M10, UBX_VALSET_LAYER_RAM)) {
            _next_message--;
        }
    }
    break;
    }
````
some how the message UBX_CFG_VALSET never get set. I turn on the debug log it show the error.
```` return _send_message(CLASS_CFG, MSG_CFG_VALGET, buf, sizeof(buf));````
This is the line that return error. I suspect that this is the GET message not the SET message. but even I change it to MSG_CFG_VALGET it still return ACK=0; 
So I decide to send the array hex to solve my problem.

peterbarker, 2024-07-31:
@andyp1per fixed this issue with https://github.com/ArduPilot/ardupilot/pull/26922

closed 2024-07-31 by peterbarker