ArduPilot/ardupilot #23548: Add support for SPARTN message format for GNSS corrections

Tal-Bass, 2023-04-19:
Ardupilot already supports RTCM messages which is a standard message format for providing GNSS corrections. SPARTN is a new standard message format for providing similar corrections. There are a couple of manufacturers of GNSS units that implement SPARTN, most notably u-blox. It would be great if AP_Periph devices had support for SPARTN. There did not seem to be a AP_Periph option under platform, so marking it as All for now.

**Platform**
[ X ] All

Reference:
https://www.spartnformat.org/

Ryanf55, 2023-04-24:
Good idea. Seems to be an improvement over the RTCM messages. Do you have any specific use cases that would be improved, or additional GPS receivers that would be supported if SPARTN were adopted?

Tal-Bass, 2023-04-24:
The u-Blox ZED-F9 series natively support PointPerfect which sends corrections via SPARTN. I think this would be a decent improvement over RTCM Messages.

ZED-F9 Series: https://www.u-blox.com/en/positioning-chips-and-modules#High-precision-GNSS
PointPerfect: https://www.u-blox.com/en/product/pointperfect

Jake-Howell, 2024-06-12:
I am also very interested in using the Spartan message format for uBlox PointPerfect. 
I'm currently trying to write a shim that converts from Spartan to RTCM3 but it would be better if this step wasn't needed!