ArduPilot/ardupilot #23556: how to increase the update frequency of imu in ardupilot?

cheslee-z, 2023-04-20:
## Bug report

I use the usb to ttl module to connect pixhawk with PC, and I want to catch pixhawk built-in imu data through mavros. Everything is fine now. But when I run the command `rostopic hz /mavros/imu/data_raw`, I find that the frequency of imu is only about 30hz, see below.
```
subscribed to [/mavros/imu/data_raw]
average rate: 25.047
	min: 0.032s max: 0.048s std dev: 0.00372s window: 25
average rate: 25.002
	min: 0.032s max: 0.048s std dev: 0.00353s window: 50
 average rate: 24.966
	min: 0.032s max: 0.048s std dev: 0.00350s window: 75
average rate: 25.002
	min: 0.032s max: 0.048s std dev: 0.00341s window: 100
average rate: 24.988
	min: 0.032s max: 0.048s std dev: 0.00330s window: 125
average rate: 24.990
	min: 0.032s max: 0.048s std dev: 0.00330s window: 150
average rate: 24.992
	min: 0.032s max: 0.048s std dev: 0.00327s window: 175
average rate: 24.988
	min: 0.032s max: 0.048s std dev: 0.00320s window: 200
average rate: 24.986
	min: 0.032s max: 0.048s std dev: 0.00313s window: 225
^Caverage rate: 24.991
	min: 0.032s max: 0.048s std dev: 0.00316s window: 239

```
I don't know why the update frequency of IMU is so low, but I want to increase its update frequency, so I used this command `rosrun mavros mavsys rate --all 100`,  but the update frequency of IMU is still very low (reference below). I don't know where the problem is, so I publish the same issue on github of mavros.

```
subscribed to [/mavros/imu/data_raw]
average rate: 34.737
	min: 0.026s max: 0.035s std dev: 0.00231s window: 34
average rate: 34.815
	min: 0.024s max: 0.035s std dev: 0.00234s window: 69
average rate: 34.485
	min: 0.024s max: 0.035s std dev: 0.00246s window: 103
average rate: 34.595
	min: 0.022s max: 0.035s std dev: 0.00254s window: 138
average rate: 34.605
	min: 0.022s max: 0.035s std dev: 0.00253s window: 173
average rate: 34.645
	min: 0.022s max: 0.035s std dev: 0.00262s window: 208
average rate: 34.707
	min: 0.022s max: 0.035s std dev: 0.00261s window: 243
average rate: 34.734
	min: 0.022s max: 0.035s std dev: 0.00269s window: 278
average rate: 34.710
	min: 0.022s max: 0.035s std dev: 0.00266s window: 312
^Caverage rate: 34.716
	min: 0.022s max: 0.035s std dev: 0.00265s window: 317
```

**Version**
ArduCopter V4.3.6 (0c5e999c)

**Platform**
[  ] All
[  ] AntennaTracker
[ √ ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quad X

**Hardware type**
pixhawk
ROS: noetic
[... 580 lines cut ...]
  - 
    level: 0
    name: "mavros: APM Hardware"
    message: "Normal"
    hardware_id: "/dev/ttyUSB0:921600"
    values: 
      - 
        key: "Core voltage"
        value: "4.793000"
      - 
        key: "I2C errors"
        value: "0"
---
```

### Check ID

```
OK. I got messages from 1:1.

---
Received 11394 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 136, 24, 152, 27, 29, 30, 33, 35, 36, 163, 165, 42, 178, 62, 193, 65, 74, 77, 111, 241, 116, 125
```

shubham-shahh, 2023-04-20:
I believe this question doesn't belong[ here](https://ardupilot.org/dev/docs/mavlink-requesting-data.html) but on the discourse or the discord server
but, here are a few ways mentioned to change rates of messages in ardupilot
NOTE: a few GCS have tendency to reset the rates, with mavproxy, you can easily shut

cheslee-z, 2023-04-21:
> I believe this question doesn't belong[ here](https://ardupilot.org/dev/docs/mavlink-requesting-data.html) but on the discourse or the discord server but, here are a few ways mentioned to change [...]

thanks. I had tried to modify those values in ardupilot, but it seems no use.

cheslee-z, 2023-04-21:
I guess maybe the delay between usb module to ttl module causes the problem.

shubham-shahh, 2023-04-21:
> thanks. I had tried to modify those values in ardupilot, but it seems no use.

did you try disabling the stream rates from GCS

cheslee-z, 2023-04-21:
> 

No, I don't know how to disable it, will doing that be fine?

shubham-shahh, 2023-04-21:
> > 
> 
> No, I don't know how to disable it, will doing that be fine?

please read through the above document carefully

closed 2023-10-21 by IamPete1