ArduPilot/ardupilot #23576: AP_DDS: Handle test hanging if dependency doesn't exist

Ryanf55, 2023-04-22:
If you didn't build all the dependnecies, I see this test hanging. Shall I create a bug?
```
ryan@ryan-B650:~/Development/ardu_test_ws$ colcon test --packages-select ardupilot_dds_tests --event-handlers=console_direct+
Starting >>> ardupilot_dds_tests
============================= test session starts ==============================
platform linux -- Python 3.10.6, pytest-6.2.5, py-1.10.0, pluggy-0.13.0
cachedir: /home/ryan/Development/ardu_test_ws/build/ardupilot_dds_tests/.pytest_cache
rootdir: /home/ryan/Development/ardu_test_ws/src/ardupilot/Tools/ros2/ardupilot_dds_tests
plugins: launch-testing-1.0.4, ament-xmllint-0.12.5, ament-lint-0.12.5, launch-testing-ros-0.19.4, ament-pep257-0.12.5, ament-copyright-0.12.5, launch-pytest-1.0.4, ament-flake8-0.12.5, cov-3.0.0, colcon-core-0.12.1
collecting ...                                    
collected 6 items                                                              

test/test_copyright.py .                                                 [ 16%]
test/test_flake8.py .                                                    [ 33%]
test/test_pep257.py .                                                    [ 50%]
test/ardupilot_dds_tests/test_navsat_msg_received.py E                   [ 66%]
test/ardupilot_dds_tests/test_time_msg_received.py E                     [ 83%]
[Processing: ardupilot_dds_tests]                   
[Processing: ardupilot_dds_tests]                           
[Processing: ardupilot_dds_tests]                             
[Processing: ardupilot_dds_tests]                             
[Processing: ardupilot_dds_tests]                             
test/ardupilot_dds_tests/test_virtual_ports.py E                         [100%]

==================================== ERRORS ====================================
_________________ ERROR at setup of test_navsat_msgs_received __________________
/opt/ros/humble/lib/python3.10/site-packages/ament_index_python/packages.py:58: in get_package_prefix
    content, package_prefix = get_resource('packages', package_name)
/opt/ros/humble/lib/python3.10/site-packages/ament_index_python/resources.py:86: in get_resource
    raise LookupError(
E   LookupError: Could not find the resource 'ardupilot_sitl' of type 'packages'

During handling of the above exception, another exception occurred:
test/ardupilot_dds_tests/conftest.py:117: in sitl_dds
    get_package_share_directory("ardupilot_sitl"),
/opt/ros/humble/lib/python3.10/site-packages/ament_index_python/packages.py:80: in get_package_share_directory
    path = os.path.join(get_package_prefix(package_name), 'share', package_name)
/opt/ros/humble/lib/python3.10/site-packages/ament_index_python/packages.py:60: in get_package_prefix
    raise PackageNotFoundError(
E   ament_index_python.packages.PackageNotFoundError: "package 'ardupilot_sitl' not found, searching: ['/home/ryan/Development/ardu_test_ws/install/ardupilot_dds_tests', '/opt/ros/humble']"
---------------------------- Captured stdout setup -----------------------------

tmpdirname: /tmp/pytest-of-ryan/pytest-1/devices0

__________________ ERROR at setup of test_time_msgs_received ___________________
/opt/ros/humble/lib/python3.10/site-packages/ament_index_python/packages.py:58: in get_package_prefix
    content, package_prefix = get_resource('packages', package_name)
/opt/ros/humble/lib/python3.10/site-packages/ament_index_python/resources.py:86: in get_resource
    raise LookupError(
E   LookupError: Could not find the resource 'ardupilot_sitl' of type 'packages'

During handling of the above exception, another exception occurred:
test/ardupilot_dds_tests/conftest.py:117: in sitl_dds
    get_package_share_directory("ardupilot_sitl"),
/opt/ros/humble/lib/python3.10/site-packages/ament_index_python/packages.py:80: in get_package_share_directory
    path = os.path.join(get_package_prefix(package_name), 'share', package_name)
/opt/ros/humble/lib/python3.10/site-packages/ament_index_python/packages.py:60: in get_package_prefix
    raise PackageNotFoundError(
E   ament_index_python.packages.PackageNotFoundError: "package 'ardupilot_sitl' not found, searching: ['/home/ryan/Development/ardu_test_ws/install/ardupilot_dds_tests', '/opt/ros/humble']"
---------------------------- Captured stdout setup -----------------------------

tmpdirname: /tmp/pytest-of-ryan/pytest-1/devices1

_____________________ ERROR at setup of test_virtual_ports _____________________
/opt/ros/humble/lib/python3.10/site-packages/launch_pytest/plugin.py:369: in run_until_complete
    loop.run_until_complete(future_like)
/usr/lib/python3.10/asyncio/base_events.py:633: in run_until_complete
    self.run_forever()
/usr/lib/python3.10/asyncio/base_events.py:600: in run_forever
    self._run_once()
/usr/lib/python3.10/asyncio/base_events.py:1860: in _run_once
    event_list = self._selector.select(timeout)
/usr/lib/python3.10/selectors.py:469: in select
    fd_event_list = self._selector.poll(timeout, max_ev)
E   KeyboardInterrupt

During handling of the above exception, another exception occurred:
/opt/ros/humble/lib/python3.10/site-packages/launch_pytest/plugin.py:127: in pytest_fixture_setup
    run_until_complete(event_loop, event.wait())
/opt/ros/humble/lib/python3.10/site-packages/launch_pytest/plugin.py:371: in run_until_complete
    if future_like.done() and not future_like.cancelled():
E   AttributeError: 'coroutine' object has no attribute 'done'
---------------------------- Captured stdout setup -----------------------------

tmpdirname: /tmp/pytest-of-ryan/pytest-1/devices2

[INFO] [launch]: All log files can be found below /home/ryan/.ros/log/2023-04-22-09-52-35-879217-ryan-B650-96148
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'ardupilot_sitl' not found, searching: ['/home/ryan/Development/ardu_test_ws/install/ardupilot_dds_tests', '/opt/ros/humble']"
[INFO] [echo-1]: process started with pid [96187]
[INFO] [echo-1]: process has finished cleanly [pid 96187]
- generated xml file: /home/ryan/Development/ardu_test_ws/build/ardupilot_dds_tests/pytest.xml -
=========================== short test summary info ============================
ERROR test/ardupilot_dds_tests/test_navsat_msg_received.py::test_navsat_msgs_received
ERROR test/ardupilot_dds_tests/test_time_msg_received.py::test_time_msgs_received
ERROR test/ardupilot_dds_tests/test_virtual_ports.py::test_virtual_ports - At...
=================== 3 passed, 3 errors in 158.84s (0:02:38) ====================
Aborted  <<< ardupilot_dds_tests [2min 39s]

Summary: 0 packages finished [2min 40s]
  1 package aborted: ardupilot_dds_tests
  1 package had test failures: ardupilot_dds_tests

```

_Originally posted by @Ryanf55 in https://github.com/ArduPilot/ardupilot/pull/23573#discussion_r1174432305_

We need to fix the dependencies, and also handle timeouts more gracefully. The test could  abort on first failure to not eat up CI.

Ryanf55, 2023-04-22:
Duplicate of https://github.com/ArduPilot/ardupilot/issues/23577

srmainwaring, 2023-04-28:
Reopening as we seemed to close both. This was first.

closed 2023-05-11 by tridge